Entware/Entware #604: build-essentials / G++

olluz, 2021-02-10:
**Describe the solution you'd like**
build-essentials or G++

Any plan on adding build-essentials, especially G++?

Thanks!

olluz, 2021-02-10:
so, it's already in there !?

zyxmon, 2021-02-11:
https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation
We do not recommend native compilation. Crosscompilation is the correct way to build binaries for Entware - https://github.com/Entware/Entware/wiki/Compile-packages-from-sources

closed 2021-02-11 by zyxmon

olluz, 2021-02-11:
Thanks a lot for the link! That is really helpful and it solved my build problems. 
I understand the reason why you recommend crosscompilation, but I need native compilation to get node-gyp (a NodeJS component) working. And it worked well with the instructions you provided above.
Thanks again