Entware/Entware #605: hevc encoder support in ffmpeg static build

hwdbk, 2021-02-11:
Could someone enable the libx265 encoder (hevc) as part of the ffmpeg build? I know, many embedded devices won't have the memory or CPU power to do a fullblown 4k HEVC encode (or it will take forever), but there are many other applications where hevc encoding is useful (smaller files, lower resolution, lower bit rates etc.). The current build (for x86_64) in entware has:

ffmpeg version 4.3.1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 8.4.0 (OpenWrt GCC 8.4.0 r1375-2df83137)
  configuration: --enable-cross-compile --cross-prefix=x86_64-openwrt-linux-gnu- --arch=x86_64 --cpu= --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --enable-lto --enable-x86asm --enable-gnutls --enable-libopus --enable-hardcoded-tables --enable-libmp3lame --enable-gpl --enable-libx264

and does do proper hevc decodes, but not encodes. The --enable-libx264 option listed above enables the AVC (h.264) encoder, but I would need the --enable-libx265 option.

Could this be added to the build, please?

- [ x ] No similar package already added and/or this one got some new killer feature(s),
- [ x ] Package is in active development (at least, not abandoned for 5 years or more),
- [ x ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2021-03-25:
This package came from OpenWrt with no any changes and we'll keep it as close to upstream as possible to simplify future updates. Sorry.

closed 2021-03-25 by ryzhovau

hwdbk, 2021-03-25:
OK, understood. It was worth the try. Cheers

LeftPinkie, 2022-12-08:
You can install the static builds from https://johnvansickle.com/ffmpeg/ which includes the build option "--enable-libx265". However you will also need to install the x265/HEVC library (libx265) as well. I have opened a package request for this lib.