Entware/Entware #609: Telegraf is an agent for collecting, processing, aggregating, and writing metrics.

wheel5up, 2021-02-15:
**Describe the solution you'd like**
I'd like to run the Telegraf agent on multiple embedded devices including a QNAP NAS and an ASUS RT-AC68U (no_hf).
https://github.com/influxdata/telegraf
I've built it on both of these devices natively and have put together the following build steps 
https://jswheeler.medium.com/monitoring-your-qnap-nas-with-telegraf-influxdb-and-grafana-e5acaba6d161
https://jswheeler.medium.com/monitoring-the-asus-rt-ac68u-with-the-telegraf-agent-grafana-and-influxdb-74356a89b1d7

- [ X] No similar package already added and/or this one got some new killer feature(s),
- [ X] Package is in active development (at least, not abandoned for 5 years or more),
- [ X] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

zyxmon, 2021-02-16:
telegraf package will be available with next Entware update - https://github.com/Entware/entware-go/commit/cc3f4d686bf862faecaf9c6fb9986e2d1e0205ea

closed 2021-02-16 by zyxmon

wheel5up, 2021-02-19:
Can you point me to a release schedule or notice? I've searched on this site, but I'm not sure when I would know there is a new release.