Entware/Entware #614: Joe package missing extra files in /opt/share/joe

VA1DER, 2021-02-26:
**Describe the bug**
[joe 4.6-1]: Syntax hjighlighting does not work on the package joe (Joe's Own Editor).  The package is missing all the files that would normally go in /usr/share/joe.  Specifically, all the charmaps, color schemes, language files, and most importantly the syntax highlighting files.  Without them syntax highlighting doesn't work, nor are other languages available.  Copying the /usr/share/joe folder from a normal installation to /opt/share/joe causes joe to function properly.

**Environment (please complete the following information):**
 - Entware feed: armv7sf-k2.6
 - Device: Netgear r6700v1
 - Firmware version Fresh Tomato 2020.8

zyxmon, 2021-02-26:
This was decided by openwrt, Contact openwrt team.

closed 2021-02-26 by zyxmon

VA1DER, 2021-02-27:
I'm curious, does Entware maintain a version of _nano_ that is different from OpenWRT?  I have noticed that Entware's _nano_ does support syntax highlighting, even though OpenWRT's version of _nano_ strips out the added files like it does for joe.

zyxmon, 2021-02-27:
Only few packages are different. We have 2000+ packages and try to keep them the same as in Openwrt. Making packages different complicates support and upgrades.

VA1DER, 2021-02-27:
> Making packages different complicates ...

... a lot of things.  I get that.  I am just wondering if nano is one of those packages that is different.  I'm am curious why nano's syntax highlighting works in Entware's package but it's not included in OpenWRT.  If you are already maintaining a different package for nano, I understand not wanting to for joe also.

For future people looking for syntax highlighting, there are two options:
1. Install nano, its syntax highlighting in Entware seems to work
2. Install joe and copy /usr/share/joe from another installation to your /opt/share/joe and joe should give syntax highlighting too.

I have a request in for the joe maintainer in OpenWRT to add a new package for joe_extras

zyxmon, 2021-02-28:
Right - nano is one of such packages:  https://github.com/Entware/entware-packages/blob/master/utils/nano/Makefile#L7