Entware/Entware #616: Make newer versions of Lighttpd available

HansEJC, 2021-03-01:
Could you please update Lighttpd. 

Version [1.4.59](https://github.com/openwrt/packages/blob/master/net/lighttpd/Makefile) comes with:
HTTP/2 enabled by default, mod_deflate zstd support, mod_ajp13 (new)

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [x ] Package is in active development (at least, not abandoned for 5 years or more),
- [ x] I'm ready to test added package on my platform [dd wrt] and get a feedback about it.

Thank you

gstrauss, 2021-03-08:
entware-packages follows openwrt and openwrt contains lighttpd 1.4.59

lighttpd 1.4.59 is already available in entware-packages
https://github.com/Entware/entware-packages/tree/master/net/lighttpd

Note: even though the next major release of OpenWRT or next major release of Entware will contain a specific version of a software package, it does not also mean that the Entware version that you run will have that version of a specific package backported or available.

lighttpd is very portable and runs on many, many systems, even older system versions.  If you know how to compile from source, you can do so yourself.  However, there may be hundreds of different distros, and those are the packagers that have to get around to updating their packages if you are waiting for an updated package for your specific distro and your specific distro release.

ryzhovau, 2021-03-25:
Done.

closed 2021-03-25 by ryzhovau