Entware/Entware #618: Node.js broken?

bunglegrind, 2021-03-04:
**Describe the bug**
Node.js is not working properly:

Latest version is installed

```
# node -v
v12.19.0
```

but if I exec node (in order to execute the REPL) or if I exec node test.js
where test.js is just a file with a console.log instruction it freezes...

**Environment:**
 - Entware feed: mipselsf-k3.4
 - Device Cisco Linksys E3000
 - Firmware version:  DD-WRT

zyxmon, 2021-03-05:
Probably DD-WRT kernel is missing some syscalls needed for some entware binaries.

zyxmon, 2021-03-05:
What is your kernel version?

bunglegrind, 2021-03-06:
# cat /proc/version
Linux version 3.10.108-d10 (root@linux) (gcc version 9.2.0 (OpenWrt GCC 9.2.0 r10890-7d542dc804) ) #2574 Tue Feb 25 14:48:51 +04 2020

bunglegrind, 2021-03-06:
Same behavior even for php-cli

`# cat /proc/version
Linux version 3.10.108-d10 (root@linux) (gcc version 9.2.0 (OpenWrt GCC 9.2.0 r10890-7d542dc804) ) #2574 Tue Feb 25 14:48:51 +04 2020
`

zyxmon, 2021-03-06:
You can make more investigations on the problem with strace. But some Entware binaries do not run on DD-WRT devices.

closed 2021-03-06 by zyxmon

bunglegrind, 2021-03-07:
ok, thanks.

BTW, is there a way to download an older version of entware packages? I wanna investigate if older versions of node still have the same issue.

bunglegrind, 2021-03-07:
oh, and php is working. It's just node.

zyxmon, 2021-03-07:
You can try entware-ng - it is old and not supported now.