Entware/Entware #625: postfix compiled without Berkeley DB support

protogen, 2021-03-21:
**Describe the bug**
postfix has been compiled without Berkeley DB support

postfix TLS requires **btree** for storing session tickets (see http://www.postfix.org/TLS_README.html).
```
smtp_tls_session_cache_database = btree:${data_directory}/smtp_tls_session_cache
smtpd_tls_session_cache_database = btree:${data_directory}/smtpd_tls_session_cache
```
postfix binary is not linked to Berkeley DB library
```
root@router:~ # opkg find 'Berkeley DB*'
libdb47 - 4.7.25.4.NC-5 - Berkeley DB library (4.7).

root@router:~ # opkg files libdb47
Package libdb47 (4.7.25.4.NC-5) is installed on root and has the following files:
/opt/lib/libdb-4.7.so
/opt/lib/libdb-4.so

root@router:~ # ldd /opt/sbin/postfix
        libresolv.so.2 => /opt/lib/libresolv.so.2 (0x2ac4f000)
        libssl.so.1.1 => /opt/lib/libssl.so.1.1 (0x2ac73000)
        libcrypto.so.1.1 => /opt/lib/libcrypto.so.1.1 (0x2aced000)
        libsasl2.so.3 => /opt/lib/libsasl2.so.3 (0x2ab92000)
        libcdb.so.1 => /opt/lib/libcdb.so.1 (0x2afe5000)
        libsqlite3.so.0 => /opt/lib/libsqlite3.so.0 (0x2aef3000)
        libpthread.so.0 => /opt/lib/libpthread.so.0 (0x2aade000)
        libdl.so.2 => /opt/lib/libdl.so.2 (0x2ab76000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0x2abb7000)
        libc.so.6 => /opt/lib/libc.so.6 (0x2afe5000)
        libz.so.1 => /opt/lib/libz.so.1 (0x2abd2000)
        libm.so.6 => /opt/lib/libm.so.6 (0x2b11f000)
        /opt/lib/ld-linux.so.3 (0x2ab1b000)
```
Ask postfix to show all supported lookup table types (btree and hash are both missing)
```
root@router:~ # postconf -m
cdb
cidr
environ
fail
inline
internal
memcache
pipemap
proxy
randmap
regexp
socketmap
sqlite
static
tcp
texthash
unionmap
unix
```

**Environment (please complete the following information):**
 - Entware feed: armv7sf-k2.6
 - Device: Asus RT-AC88U router
 - Firmware version: asuswrt-merlin 386.1_2

ryzhovau, 2021-04-06:
We trying to keep packages lightweight as possible. Please, take a look at another features which you concern as most useful, I'll enable them too:
```
CONFIG_POSTFIX_TLS=y
CONFIG_POSTFIX_SASL=y
# CONFIG_POSTFIX_LDAP is not set
# CONFIG_POSTFIX_DB is not set
CONFIG_POSTFIX_CDB=y
CONFIG_POSTFIX_SQLITE=y
# CONFIG_POSTFIX_MYSQL is not set
# CONFIG_POSTFIX_PGSQL is not set
# CONFIG_POSTFIX_PCRE is not set
# CONFIG_POSTFIX_EAI is not set
```

protogen, 2021-04-07:
**TL;DR**
I believe Berkeley DB is the only additional feature that needs to be enabled.  All other features (LDAP, MYSQL, PGSQL, PCRE, EAI) have equivalent alternatives or have a low probability of use and can therefore remain disabled.

```
CONFIG_POSTFIX_TLS=y
CONFIG_POSTFIX_SASL=y
# CONFIG_POSTFIX_LDAP is not set
CONFIG_POSTFIX_DB=y
CONFIG_POSTFIX_CDB=y
CONFIG_POSTFIX_SQLITE=y
# CONFIG_POSTFIX_MYSQL is not set
# CONFIG_POSTFIX_PGSQL is not set
# CONFIG_POSTFIX_PCRE is not set
# CONFIG_POSTFIX_EAI is not set
```

---

**Feature Review**
I've read all the documentation (and many discussion threads) on each of the additional features to better understand their use cases and prerequisites.  My summary of each is below.

1. **DB (Berkeley DB)**
The use cases for lookup table types `btree` and `hash` have no other alternatives.  In the situation where Berkeley DB is not available, an administrator is unable to achieve the outcome they desire.
**Conclusion:** feature DB should be enabled

2. **LDAP, MYSQL and PGSQL**
The use cases for lookup table types `ldap`, `mysql` and `pgsql` can also be accomplished with either `hash` or `sqlite`.  In the situation where `ldap`, `mysql` and `pgsql` are not available, an administrator can still achieve the outcome they desire using either `btree`, `hash` or `sqlite`.
**Conclusion:** features LDAP, MYSQL and PGSQL can remain disabled

3. **PCRE**
The use cases for lookup table type `pcre` can also be accomplished with `hash`.  Granted, a `hash` lookup table would be larger and less efficient than `pcre`, however it would still be functionally equivalent.
**Conclusion:** feature PCRE can remain disabled

4. **EAI**
The use case for EAI (Email Address Internationalization - basically UTF-8 support) has low probability.  To properly enable SMTPUTF8 the entire SMTP chain must support it.  This would include many individual pieces of software, not just postfix.
**Conclusion:** feature EAI can remain disabled

ryzhovau, 2021-04-07:
Thank you for information, it's really appreciated.
Postfix has a quite ancient build scripts, which isn't handy for cross-compilation. Please, feel free to report any other issues you've found.

closed 2021-04-07 by ryzhovau