Entware/Entware #627: htop: 3.0.3 and up does not show tasks anymore (3.0.2 and prior were ok)

bolemo, 2021-03-27:
**Environment**: cortex a15, Netgear R7800, Entware (Voxel’s feed)

**Description**:
I am currently using htop 3.0.1-1 through Entware 3x for cortex a15 (router Netgear R7800).

While 3.0.1-1 works flawlessly (as did prior versions), after I upgraded htop to version 3.0.5-1 (using the opkg update the upgrade process), I noticed it is not working properly anymore.
Number of tasks reported is 0, and no tasks are shown at all in the list.
Other info (CPU, Mem, Swp, etc...) seems to be fine, and regular top works normal.

When downgraded back to 3.0.1-1, it worked again instantly (no need to restart).

I then compiled from tarballs 3.0.2 and 3.0.3 (from Entware):
3.0.2 works fine.
3.0.3 presents the same issue (tasks missing and numbered to 0).

**Useful/related links**:
I initially reported the bug here: https://github.com/openwrt/packages/issues/15167

Similar bug here: https://github.com/htop-dev/htop/issues/534#issuecomment-783322981
I tried the workaround mentioned in there, but /proc is not a symlink for me, so it does not solve the problem. Interesting though that the same issue appears starting with the same version.

Topic where I mention the bug is SNB forum: https://www.snbforums.com/threads/entware.65515/page-2#post-674495

zyxmon, 2021-03-27:
Voxel feed has now only htop 3.0.1
My guess that `ac_cv_func_openat=no` will fix the build for you.

PS I have 10 devices at home running Entware. This bug was only on one of the device quite different from yours (armv5 NAS). The configuration above solve the problem.

bolemo, 2021-03-27:
> Voxel feed has now only htop 3.0.1
> My guess that `ac_cv_func_openat=no` will fix the build for you.
> 
> PS I have 10 devices at home running Entware. This bug was only on one of the devices and ot is quite different from yours (armv5 NAS). The configuration above solved the problem.

Voxel feed last release has htop 3.0.5.
`ac_cv_func_openat=no` does fix it, **thank you!**
Should it be fixed in Entware, or only Voxel’s feed?

zyxmon, 2021-03-27:
It is already fixed in standard Entware feeds. Voxel will probably fix it in his feed later.

closed 2021-03-27 by zyxmon