Entware/Entware #628: Dlink 340L - Nearest Suitable Guide

marcoczen, 2021-03-29:
Guys,

I have a Dlink 340L with the following specs;

CPU Marvell Armada 370 88F6707-A1 C120 @ 1.2GHz
RAM 512 MB Hynix DDR3 (x2)
Flash 128 MB Hynix
Ethernet Marvell 88E1518 Gigabit Ethernet (x2)
USB 3.0 Etron Tech EJ168A
SATA Marvell 88SE9215 4 port PCIe 2.0 SATA 6Gb/s and PATA Host Controller

I have seen the  wiki at  https://github.com/Entware/Entware/wiki but unsure as to which guide will be closest to a Dlink setup ( from the specs above ).

I do have ffp running ( mutt, ssh , rsync ) but the packages seem outdated ? Its dead right ?
So i rather move on to something that is properly supported like Entware.

ryzhovau, 2021-03-29:
So, the question is how to install Entware. A few basic requirements:
* a writeable, clean `/opt` folder,
* appropriate CPU architecture,
* ability to run `/opt/etc/init.d/rc.unslung` after boot.

Please, show us `cat /proc/cpuinfo` to find out which Entware feed better to use.

marcoczen, 2021-03-29:
Hi ...

# cat /proc/cpuinfo
Processor	: Marvell PJ4Bv7 Processor rev 1 (v7l)
BogoMIPS	: 1196.85
Features	: swp half thumb fastmult vfp edsp vfpv3 vfpv3d16 tls 
CPU implementer	: 0x56
CPU architecture: 7
CPU variant	: 0x1
CPU part	: 0x581
CPU revision	: 1

Hardware	: Marvell Armada-370
Revision	: 0000
Serial		: 0000000000000000

> ability to run /opt/etc/init.d/rc.unslung after boot.

How can i test that ?

ryzhovau, 2021-03-29:
> Processor : Marvell PJ4Bv7 Processor rev 1 (v7l)

It's ARMv7, so `https://bin.entware.net/armv7sf-k2.6/installer/generic.sh` or `https://bin.entware.net/armv7sf-k3.2/installer/alternative.sh` can be used, depends on kernel version (see `uname -a`).

> How can i test that ?

You can't:) You should provided that. Better to ask details on D-Link users forum. Without that option you will install Entware, but  no additional services will be started automatically.

> My current dir structure is as below ...

`/opt` is present, take a look at it's content (`ls -la /opt`).

marcoczen, 2021-03-29:
Hi .. Replies Below ...

> > Processor : Marvell PJ4Bv7 Processor rev 1 (v7l)
> 
> It's ARMv7, so `https://bin.entware.net/armv7sf-k2.6/installer/generic.sh` or `https://bin.entware.net/armv7sf-k3.2/installer/alternative.sh` can be used, depends on kernel version (see `uname -a`).
> 

Ok. Thanks.

> 
> `/opt` is present, take a look at it's content (`ls -la /opt`).

    # pwd
       /opt
    # ls -al
      total 2
      drwxr-xr-x  2 root root 1024 Jan  1  1970 .
      drwxr-xr-x 18 root root 1024 Mar  4 00:47 ..
      lrwxrwxrwx  1 root root   18 Jan  1  1970 firefly -> /usr/local/firefly
      lrwxrwxrwx  1 root root   27 Jan  1  1970 perl5.10 -> /usr/local/modules/perl5.10

ryzhovau, 2021-03-29:
Both `firefly` and `perl5.10` doesn't conflict with Entware deployment. 
If you'll have to remove existing Entware installation in future, just clean everything in `/opt`, except those `firefly` and `perl5.10`.

> Linux DLINK-340L 3.2.40 #1 Fri May 9 16:33:59 CST 2014 armv7l GNU/Linux

`https://bin.entware.net/armv7sf-k3.2/installer/alternative.sh` is the best option to install Entware.

The last question is autostart. You may try to deploy Entware, use any utilities or daemons started manually. Good luck!

I'm closing this Issue for now.

closed 2021-03-29 by ryzhovau

marcoczen, 2021-03-29:
> Both `firefly` and `perl5.10` doesn't conflict with Entware deployment.

Ok. Noted.

> If you'll have to remove existing Entware installation in future, just clean everything in `/opt`, except those `firefly` and `perl5.10`.
> 

Noted.

> > Linux DLINK-340L 3.2.40 #1 Fri May 9 16:33:59 CST 2014 armv7l GNU/Linux
> 
> `https://bin.entware.net/armv7sf-k3.2/installer/alternative.sh` is the best option to install Entware.

Ok. 
> 
> The last question is autostart. You may try to deploy Entware, use any utilities or daemons started manually. Good luck!

So i can use Entware except that if i reboot the nas, i would have to start services manually ?
Does this mean that i have to manually start up cron ?
Which will then run my rsync as per the cron entry ?

> 
> I'm closing this Issue for now.

Thanks.

ryzhovau, 2021-03-29:
I'm not ready to provide personal support, sorry.

Neustradamus, 2026-01-02:
@marcoczen: Can you publish all D-Link DNS-340L details?

Linked to:
- https://datasheet.octopart.com/88F6W11-A1-BMI2C080-Marvell-datasheet-180622313.pdf
- https://wikidevi.wi-cat.ru/D-Link_DNS_series
- https://techinfodepot.shoutwiki.com/wiki/D-Link_DNS_series

There are already other D-Link DNS models.

Note: OpenWrt supports DNS-320L and DNS-325:
- https://github.com/openwrt/openwrt/pull/17924