Entware/Entware #629: NextDNS package seems to ignore the configuration file

Lichtenshtein, 2021-04-01:
**Describe the bug**
[package name]: nextdns - v1.10.1-1

Can't change the default port no matter what i do.

nextdns install
then

```
/opt/home/admin # S09nextdns start
 Starting nextdns...              done.
/opt/home/admin # S09nextdns check
 Checking nextdns...              dead.
/opt/home/admin # nextdns start
/opt/home/admin # nextdns status
running
/opt/home/admin # nextdns config
bogus-priv true
setup-router false
control /var/run/nextdns.sock
cache-size 5MB
detect-captive-portals false
use-hosts true
log-queries false
max-ttl 5s
report-client-info false
listen localhost:6353
cache-max-age 0s
discovery-dns
hardened-privacy false
timeout 5s
auto-activate false
/opt/home/admin # nextdns run
INFO: 13:56:05 Starting NextDNS dev/linux on localhost:53
INFO: 13:56:05 Listening on TCP/127.0.0.1:53
INFO: 13:56:05 Listening on UDP/127.0.0.1:53
ERROR: 13:56:05 Startup failed: proxy: tcp: listen tcp 127.0.0.1:53: bind: address already in use
Error: proxy: tcp: listen tcp 127.0.0.1:53: bind: address already in use
```
Don't know if it's the NextDNS issue or Entware bug
https://github.com/nextdns/nextdns/issues/44

`setup-router ignores listen`

Maybe it autodetects that it's running on router and uses 'setup-router true' by default which ignores the listen line in config file? Also i can't delete setup-router line from file, NextDNS recreates it automatically.

**Environment (please complete the following information):**
 - Entware feed: [mipselsf-k3.4]
 - Device [Xiaomi Mi-MINI router]
 - Firmware version [Padavan, Entware]

zyxmon, 2021-04-01:
I am afraid you are the first user that I know that uses Entware nextdns package. Entware version of nextdns is the same as the one form nextdns releases page. The only difference is go binaries compiled for Entware use some files (certificates for example) from /opt folder. Hera is a patch we use - https://github.com/Entware/Entware/blob/master/tools/go-src/patches/500-entware.patch
AFAIK Entware developers do not use nextdns - lets wait some of the users will help you.
PS It  was a surprise that nextdns project has special installation  - https://github.com/nextdns/nextdns/tree/master/host/service/entware
PPS In case nextdns package is good - we will fix it here - https://github.com/nextdns/nextdns/blob/master/config/ctl_unix.go#L5 to use /opt folder. In case it does not work - we will simply remove it from repo.

Lichtenshtein, 2021-04-04:
I see that this topic was mentioned here
https://github.com/nextdns/nextdns/issues/44
Have the dev been notified about it somewhere automatically?
Or shall i write about an issue there too?

zyxmon, 2021-04-04:
AFAIK /etc folder is writable on Padavan firmware. You can create a link /etc/ssl->/opt/etc/ssl and try nextdns binary from https://github.com/nextdns/nextdns/releases to investigate if it is nexdns issue or entware bug.

Lichtenshtein, 2021-04-04:
> AFAIK /etc folder is writable on Padavan firmware. You can create a link /etc/ssl->/opt/etc/ssl

I already have /opt/etc/ssl folder with cert.pem in it (don't know where from). Do i need to do that?

> and try nextdns binary from https://github.com/nextdns/nextdns/releases to investigate if it is nexdns issue or entware bug.

Yes i can reproduce the same issue with that official binary
https://github.com/nextdns/nextdns/releases/download/v1.11.0/nextdns_1.11.0_linux_mipsle_softfloat.tar.gz

Also nextdns shows 'running' with 'nextdns status' command when it's not.

zyxmon, 2021-04-04:
>> Yes i can reproduce the same issue with that official binary
Please wright about it on nextdns official issues page.

closed 2021-04-04 by zyxmon

Alominer, 2021-04-18:
Solution to the problem:
Insert to /opt/etc/init.d/S09nextdns 
`ARGS="run -config-file /opt/etc/nextdns.conf"`

/opt/etc/nextdns.conf
```
discovery-dns
auto-activate false
listen 192.168.1.1:5453 <- listen LAN interface otherwise, clients are not displayed on the site
cache-size 10MB
setup-router false
cache-max-age 0s
max-ttl 0s
hardened-privacy false
bogus-priv true
use-hosts true
timeout 5s
log-queries false
report-client-info true
detect-captive-portals false
control /var/run/nextdns.sock
config yourID
```
In dnsmasq.conf add the following lines
```
no-resolv
bogus-priv
domain-needed
strict-order
server=192.168.1.1#5453
add-mac
add-subnet=32,128
```