Entware/Entware #631: Repository download speed

falcon4fun, 2021-04-06:
Do u have any problems with one of servers? Because i'm getting random speed from some KB/s and hangout to MB/s
It's like server 104.21.91.83 having speed issues. So I cant simply `opkg upgrade` because some packets get "returned 1" and some get stuck infinitely.

Temp solution:
Mapped to 172.67.212.134 using dnsmasq option:
address=/bin.entware.net/172.67.212.134
and updated without any hangs, stucks and issues

Log from router trying to download the same packet multiple times
https://paste.ee/p/6nH1T

ryzhovau, 2021-04-07:
Looks like some Cloudflare issue. I'll keep this issue open to monitor other users feedback.

ryzhovau, 2021-07-06:
LGTM

closed 2021-07-06 by ryzhovau