Entware/Entware #638: pciutils

dajiahao, 2021-04-16:
possible bug on installing pciutils; tried on x64 and armv7 qnap architecture

# opkg install pciutils
Installing pciutils (3.7.0-2) to root...
Downloading http://bin.entware.net/x64-k3.2/pciutils_3.7.0-2_x64-3.2.ipk
Configuring pciutils.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  267k  100  267k    0     0   912k      0 --:--:-- --:--:-- --:--:--  917k
zcat: no gzip/bzip2 magic
update-pciids: missing class info, probably truncated file
rm: can't remove 'pci.ids.gz.old': No such file or directory

# opkg install pciutils
Installing pciutils (3.7.0-2) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/pciutils_3.7.0-2_armv7-3.2.ipk
Configuring pciutils.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  267k  100  267k    0     0  1184k      0 --:--:-- --:--:-- --:--:-- 1448k
zcat: Invalid magic
update-pciids: missing class info, probably truncated file
rm: cannot remove `pci.ids.gz.old': No such file or directory

zyxmon, 2021-04-16:
This is not a severe error. To fix it - edit /opt/sbin/update-pciids and change one of the first lines to `PCI_COMPRESSED_IDS=1`. Run script `/opt/lib/opkg/info//pciutils.postinst` to download pci ids.

dajiahao, 2021-04-16:
thanks, works very well

zyxmon, 2021-04-17:
In fact the error string comes from postinstallation script that is obsolete now.
package works correctly without any manual editing. It is fixed now via https://github.com/Entware/entware-packages/commit/6ce75de7021e325ad52c08a7a1730c1ece9562d4

closed 2021-04-17 by zyxmon

ryzhovau, 2021-04-17:
@dajiahao, thanks for report.