Entware/Entware #639: Entware repo contains BIND 9.17.10 - this is an unstable development release

protogen, 2021-04-17:
**Describe the bug**
Entware repo currently contains BIND 9.17.10 - the entire 9.17 branch is an **unstable** development release.
The latest **stable** version of BIND is 9.16.13 (see references below).  Please update the Entware repo to use the latest stable version.

https://bind9.readthedocs.io/en/latest/notes.html#introduction
https://www.isc.org/download/   (click on **BIND 9** half way down)
https://en.wikipedia.org/wiki/BIND

**Environment (please complete the following information):**
 - Entware feed: armv7sf-k2.6
 - Device Asus RT-AC88U router
 - Firmware version: asuswrt-merlin 386.1_2

ryzhovau, 2021-04-18:
This package came from [upstream](https://github.com/openwrt/packages/tree/master/net/bind) with no changes and we'll keep it in this state to simplify further updates.

closed 2021-04-18 by ryzhovau

protogen, 2021-04-18:
BIND 9.17.10 does not work properly.  Upon start-up, it immediately logs numerous errors and fails to function 100% (some functions work, some do not).

What is the path forward to get this fixed?  Should I raise an issue with [openwrt](https://github.com/openwrt)?

protogen, 2021-04-18:
@ryzhovau
I raised an [issue](https://github.com/openwrt/packages/issues/15447) with OpenWRT. 

Their response implies Entware is pulling from a **development** branch of OpenWRT (that's why you are getting a development branch of BIND).  They say their stable branch of OpenWRT will contain the latest stable branch of BIND.

Can you comment?

ryzhovau, 2021-04-19:
Sure, that's our goal: we sync most of our packages with [OpenWrt dev feed](https://github.com/openwrt/packages) to keep it updated. It allows to keep repo in fresh state with minimal efforts of couple of maintainers. 

I understand that isn't what you expected, sorry. We can't satisfy all requests, just most of them.

protogen, 2021-04-19:
Thank you for clarifying.  I wasn't aware you were syncing with OpenWrt dev feed.  I can now see the problem is not Entware - the problem is my own ignorance!

I apologise for the confusion, Alexander.  I hope I didn't waste too much of your time. :-)