Entware/Entware #640: Mosh Server needs older version of protobuf

camkidd, 2021-04-21:
Mosh server does not run due to more recent version of libprotobuf being installed. Not sure if this is entware compiling or upstream issue.
```
# mosh-server
mosh-server: error while loading shared libraries: libprotobuf.so.3.14.0.0: cannot open shared object file: No such file or directory.
# opkg info mosh-server | grep Version
Version: 1.3.2-2
# opkg info protobuf | grep Version
Version: 3.15.6-2
```
Downgrading to protobuf_3.14.0-2_armv7-3.2.ipk (from https://bin.entware.net/armv7sf-k3.2/archive/protobuf_3.14.0-2_armv7-3.2.ipk) does work.

Thanks

 - Entware feed: armv7
 - Device Asus RT-AX3000U router
 - Firmware version asuswrt-merlin 386.2

ryzhovau, 2021-04-21:
Same-old opkg issue, it's to simple to track down this kind of dependences. Please, reinstall mosh. It surely linked against new protobuf library.
https://github.com/Entware/Entware/wiki#note-about-packages-upgrade

closed 2021-04-21 by ryzhovau

camkidd, 2021-04-21:
Thankyou. Thought I had reinstalled but guess not.