Entware/Entware #641: qbittorent will not install on a Western Digital DL4100 (OS5) NAS

Myron-S, 2021-04-24:
After _opkg upgrade_ qbittorrent failed to load.  I've tried to uninstall and re-install qbittorrent and have been greeted with ...

```
root@NAS etc # opkg install qbittorrent
Installing qbittorrent (4.3.4.1-1) to root...
Downloading http://bin.entware.net/x64-k3.2/qbittorrent_4.3.4.1-1_x64-3.2.ipk
Installing libtorrent-rasterbar (2.0.1-1) to root...
Collected errors:
 * opkg_download_pkg: Package libtorrent-rasterbar is not available from any configured src.
 * opkg_install_pkg: Failed to download libtorrent-rasterbar. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package qbittorrent.
root@NAS etc #
```

```
root@NAS etc # uname -a
Linux NAS 4.14.22 #1 SMP Mon Dec 21 03:06:55 UTC 2020 Build-26 x86_64 GNU/Linux
root@NAS etc #
```

zyxmon, 2021-04-25:
We have downgraded libtorrent-rasterbar from 2.x to 1.2.x in the repo for more stability.
Opkg cannot handle this situation, You are to fix it manually.
In case libtorrent-rasterbar is shown installed (opkg list-installed), run
```
opkg remove --force-depends libtorrent-rasterbar
opkg install libtorrent-rasterbar
```
Sometimes installed packages list is not correct, Edit the file `/opt/lib/opkg/status` and remove `libtorrent-rasterbar` section. After it
`opkg install libtorrent-rasterbar`
Please write what have helped you.
**All** PS similar situation is for qbittorrent package.

Myron-S, 2021-04-25:
Thank-you.  I had to remove the `libtorrent-rasterbar` section.

closed 2021-04-25 by Myron-S