Entware/Entware #644: ffmpeg 4.3.2 broken by upgrade

0ddb0d, 2021-04-27:
After opkg updating/upgrading all my packages ffmpeg no longer worked. Error message was

`ffmpeg: error while loading shared libraries: libx264.so.161: cannot open shared object file: No such file or directory`

The cause appeared to be the wrong version of libx264

```
[~] # ldd /opt/bin/ffmpeg
        linux-vdso.so.1 (0x00007fff94459000)
        libc.so.6 => /opt/lib/libc.so.6 (0x00007f81a332c000)
        libavdevice.so.58 => /opt/lib/libavdevice.so.58 (0x00007f81a331b000)
        libavfilter.so.7 => /opt/lib/libavfilter.so.7 (0x00007f81a2f80000)
        libavformat.so.58 => /opt/lib/libavformat.so.58 (0x00007f81a2d10000)
        libavcodec.so.58 => /opt/lib/libavcodec.so.58 (0x00007f81a1881000)
        libpostproc.so.55 => /opt/lib/libpostproc.so.55 (0x00007f81a1864000)
        libswresample.so.3 => /opt/lib/libswresample.so.3 (0x00007f81a1844000)
        libswscale.so.5 => /opt/lib/libswscale.so.5 (0x00007f81a17b6000)
        libavutil.so.56 => /opt/lib/libavutil.so.56 (0x00007f81a1526000)
        libm.so.6 => /opt/lib/libm.so.6 (0x00007f81a1393000)
        libpthread.so.0 => /opt/lib/libpthread.so.0 (0x00007f81a1373000)
        /opt/lib/ld-linux-x86-64.so.2 (0x00007f81a34e2000)
        libasound.so.2 => /opt/lib/libasound.so.2 (0x00007f81a1290000)
        libbz2.so.1.0 => /opt/lib/libbz2.so.1.0 (0x00007f81a127b000)
        libz.so.1 => /opt/lib/libz.so.1 (0x00007f81a1261000)
        libgnutls.so.30 => /opt/lib/libgnutls.so.30 (0x00007f81a1049000)
        libmp3lame.so.0 => /opt/lib/libmp3lame.so.0 (0x00007f81a0fcf000)
        libopus.so.0 => /opt/lib/libopus.so.0 (0x00007f81a0f74000)
        libx264.so.161 => /opt/lib/libx264.so.161 (0x00007f81a0ce3000)
        libdl.so.2 => /opt/lib/libdl.so.2 (0x00007f81a0cdc000)
        librt.so.1 => /opt/lib/librt.so.1 (0x00007f81a0cd2000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0x00007f81a0cb8000)
        libnettle.so.8 => /opt/lib/libnettle.so.8 (0x00007f81a0c78000)
        libhogweed.so.6 => /opt/lib/libhogweed.so.6 (0x00007f81a0c30000)
        libgmp.so.10 => /opt/lib/libgmp.so.10 (0x00007f81a0bb8000)
        libssp.so.0 => /opt/lib/libssp.so.0 (0x00007f81a0bb1000)
```

I tried updating/upgrading again with opkg but found that libx264.so.157 was still present. Removing ffmpeg, libffmpeg-full and libx264 then reinstalling ffmpeg resolved the problem. I don't know what went wrong as there were no errors during the upgrade and no other packages are currently causing problems.

**Environment (please complete the following information):**
 - Entware feed: x64-k3.2
 - Device: QNAP TS-869L
 - Firmware version QTS 4.3.4.1652

Thank you for continuing to maintain Entware, my NAS wouldn't be half as useful without it.

ryzhovau, 2021-04-27:
`opkg` too simple to track down changed dependences, sorry.

https://github.com/Entware/Entware/wiki#note-about-packages-upgrade

closed 2021-04-27 by ryzhovau

zyxmon, 2021-04-27:
In case the article from wiki does not help -
1. Edit `/opt/lib/opkg/status` file and delete libx264 section (you can use nano or another editor from entware)
2. run `opkg install libx264`