Entware/Entware #646: Add jsh portable javascript scripting language - created by the fossil/sqlite3 guy.

timoc, 2021-04-29:
https://jsish.org/fossil/jsi/doc/ckout/README.md
Support for jsish scripting.

Jsi is a javascript-ish interpreter with builtin websocket-server, sqlite and C-extensibility. It is an all in one, cross platform scripting tool. Essentially an embedded webserver with sql and mysql integration.  

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2021-04-30:
Not sure we want another (6th?) web server in repo, sorry.

closed 2021-04-30 by ryzhovau

timoc, 2021-05-01:
The cross platform scripting with node like web technology built in is how i would prefer to frame it. i added the webserver part, because it is a very interesting use case (2 files == web application) for embedded devices.