Entware/Entware #649: [ffmpeg] incompatibility with some merging operations

GregThib, 2021-05-02:
**The bug**
[ffmpeg]: I frequently use ffmpeg, directly or indirectly. For instance, I used it from (with) youtube-dl application. And, since my last upgrade, I do received the following message when youtube-dl try to merge sound and video files into a mkv container:
`ERROR: ffmpeg: error while loading shared libraries: libx264.so.161: cannot open shared object file: No such file or directory`

**Environment:**
 - Entware feed: x64-k3.2
 - Device: Synology (DS1512+, based on a Intel x64 CPU)
 - Firmware version: DSM 6.2.4

**Workaround (temporary fix)**
- download an outdated version of ffmpeg and dependencies (before 4.3.1-1):
    `wget https://bin.entware.net/x64-k3.2/archive/{ffmpeg,libffmpeg-full,ffprobe}_4.2.2-4_x64-3.2.ipk`
- install them with --force-upgrade:
    `sudo opkg install --force-downgrade {ffmpeg,libffmpeg-full,ffprobe}_4.2.2-4_x64-3.2.ipk`
- hold them to await a new stable and working version:
    `sudo opkg flag hold {ffmpeg,libffmpeg-full,ffprobe}`

**more info**
_sorry, but I didn't understand effectively the error, and I'm not able to understand further due to my competencies :( _

zyxmon, 2021-05-02:
duplicate - https://github.com/Entware/Entware/issues/644

closed 2021-05-02 by zyxmon

GregThib, 2021-05-02:
Oooops, sorry, I forgot to search in the closed ones before posting…