Entware/Entware #650: Redis не работает.

supersh86, 2021-05-04:
Всем привет, при установки redis-server и запуску его через конфиг по умолчанию

`redis-server /opt/etc/redis.conf `
происходит ошибка 

`8415:C 04 May 2021 15:14:29.880 # Can't chdir to '/opt/var/lib/redis': No such file or directory`

В самой папке /opt/var/lib/redis после установки ничего не появилось. 
Папка расшарена, nginx и mysql там спокойно встал. 

Использую Entware для android box armv7

zyxmon, 2021-05-04:
Android не равно linux. Очень часто в android не хватает системных файлов linux. Используйте strace и другие средства отладки. Поищите решение в google.

zyxmon, 2021-05-04:
После создания папки /opt/var/lib/redis у меня запускается
```
# redis-server /opt/etc/redis.conf
# ps | grep redis
 5936 root     14628 S    /mnt/ext/opt/netmgr/util/redis/redis-server *:0
19858 root     33268 SN   redis-server 127.0.0.1:6379
```

supersh86, 2021-05-04:
> После создания папки /opt/var/lib/redis у меня запускается
> 
> ```
> # redis-server /opt/etc/redis.conf
> # ps | grep redis
> 19858 root     33268 SN   redis-server 127.0.0.1:6379
> ```

Спасибо заработало. В любом случае проблема не в андройде, а в отсутствии папки.

closed 2021-05-05 by zyxmon