Entware/Entware #652: tang package installation fails

kinafu, 2021-05-08:
The installation of the package `tang` fails for the same reason as mentioned in #560. The issue has been closed, because tang has been added to the repository. But the installation of the package fails.

```sh
opkg update
opkg upgrade
opkg install tang
```
The `opkg install tang` command returns
```
Installing tang (8-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/tang_8-1_armv7-2.6.ipk
Configuring tang.
//opt/lib/opkg/info/tang.postinst: line 15: /opt/libexec/tangd-update: not found
```

 - Entware feed: armv7-2.6

kinafu, 2021-06-19:
The issue has been addressed by the openwrt package maintainer (thanks @neheb), see https://github.com/openwrt/packages/issues/14692.  
The changes to the tang package have been [merged](https://github.com/openwrt/packages/commit/d49c79b76b7c7659e063290d1bc7d7fad626780a) into the `openwrt` master branch.
To me it seems like the changes could be synced with `entware`. But I do not know, whether or not the changes to `tang` in `openwrt` have been tested.

zyxmon, 2021-06-19:
@kinafu - plz check if http://test.entware.net/armv7sf-k2.6/tang_10-1_armv7-2.6.ipk works

ryzhovau, 2021-07-06:
No response

closed 2021-07-06 by ryzhovau

kinafu, 2021-07-06:
I accidentally responded here (https://github.com/Entware/Entware/issues/560) to your provided package.

Until now I haven't had the chance to get physical access to the router I use for testing.

Should the issue be further discussed here or in the issue linked above?

zyxmon, 2021-07-06:
Discuss it here.