Entware/Entware #653: [Question] Where can I find the source code of all packages?

JackMerlin, 2021-05-15:
I checked these repositories, but the source code of many packages was missing. 
https://github.com/Entware/rtndev
https://github.com/Entware/entware-oldpackages-ports
https://github.com/Entware/entware-ng-rtndev

For example: source code of acl, opkg...

I can't find the source code for many of the latest packages. please help me.

zyxmon, 2021-05-15:
source coude is downloaded to dl folder when you build package. Download uses PKG_SOURCE as a part or download URL.

closed 2021-05-15 by zyxmon

JackMerlin, 2021-05-15:
> 
> 
> source coude is downloaded to dl folder when you build package. Download uses PKG_SOURCE as a part or download URL.

Thanks, but do you mean that I need to find the source of the developer for each package instead of all the package source code hosted by entware?

ryzhovau, 2021-05-16:
Build system gets instructions from package Makefile, you may see `opkg` Makefile [here](https://github.com/Entware/Entware/blob/master/package/system/opkg/Makefile). It performs set of actions for every package: 

* Download source code to `dl` folder, opkg source is https://git.openwrt.org/project/opkg-lede.git
* Unpacks sources and applies patches if needed (Prepare stage),
* Configures and then, compiles source code,
* Pack binaries into *.ipk file.

So, Entware itself doesn't store any package sources on Github. If you want to update `opkg`, you have to…

*  `git clone https://git.openwrt.org/project/opkg-lede.git`, 
* see what's new from commit we use, 
* adapt new features you want using your own patches,
* compile and test new version of `opkg`.

As long as we use OpenWrt Buildroot, you may refer to [documentation ](https://openwrt.org/docs/guide-developer/build-system/use-buildsystem) for details.

JackMerlin, 2021-05-17:
> Build system gets instructions from package Makefile, you may see `opkg` Makefile [here](https://github.com/Entware/Entware/blob/master/package/system/opkg/Makefile). It performs set of actions for [...]

Sorry, I don't know much about how to work. Thank you very much for the friendly reply to the novice.