Entware/Entware #661: Problems with "Using Pip"

bbua97, 2021-06-08:
I just want to get run a python script on my QNAP NAS TS219P II. I´m not very experiences with programming languages but i have an experienced friend and even he got stucked... I followed all the steps:
- Installed Entware
- Installed gcc (not sure if i did right)
- Installed Python (i´m stucking at)

The script i want to run is: 
https://github.com/Tkd-Alex/Twitch-Channel-Points-Miner-v2

Everytime i try to install the requirements.txt i get a loooot of red lines in my putty. It is telling me always "ERROR: Command errored out with exit status 1: ..." No package libffi found and also no working compiler found. As you see i´m not really knowing what I´m doing but maybe you can help me out. 

Kind regards

bbua97, 2021-06-09:
> I just want to get run a python script on my QNAP NAS TS219P II. I´m not very experiences with programming languages but i have an experienced friend and even he got stucked... I followed all the [...]

Using Python
Sometimes you may encounter some errors during compilation. For example when you use pip to install cffi the libffi library will not be found. Entware (Openwrt in fact) does not install libffi.so symlink in /opt/lib. There is only symlink libffi.so.6 there. You can create missing link manually or fix the installation files. This can be done as follows. First download cffi archive and unpack to some directory. Short investigation shows that setup.py has a line libraries = ['ffi'] with the missing library. Edit it, and change to libraries = [':libffi.so.6']. Now run python setyp.py build.

I think this is the part I´m stucking. Can you please tell me where and how i can find this "setup.py" file?

ryzhovau, 2021-06-09:
We can't provide personal support, sorry.

closed 2021-06-09 by ryzhovau