Entware/Entware #663: Strangeness with NTPD 4.2.8p15-3 on ARMv7

jbaker6953, 2021-06-09:
I'm trying to run NTPD on an Asus RT-AC88U router, but experiencing some odd issue. I don't know how to explain it, so I'll show it. These are two ntpq -p queries less than 5 seconds apart:

`admin@RT-AC88U-B1E8:/tmp/mnt/entware/entware/sbin# ntpq -p
     remote           refid      st t when poll reach   delay   offset  jitter
==============================================================================
x.ns.gin.ntt.ne 249.224.99.213   2 u  51y   64    3    0.000   +0.000   0.000
ntp.seby.io     17.253.66.253    2 u   48   64    3  153.602   +7.114   0.002
ntp1.ds.network 92.21.53.217     2 u   48   64    3  239.395  +62.005 8113392
pve03.as24220.n 220.158.215.20   3 u   49   64    3  165.027  +13.243   0.002
admin@RT-AC88U-B1E8:/tmp/mnt/entware/entware/sbin# ntpq -p
     remote           refid      st t when poll reach   delay   offset  jitter
==============================================================================
x.ns.gin.ntt.ne 249.224.99.213   2 u  51y   64    7    0.000   +0.000   0.000
ntp.seby.io     17.253.66.253    2 u  51y   64    7  1622678  +811339   0.002
ntp1.ds.network 92.21.53.217     2 u  51y   64    7  1622678  +811339 8113393
pve03.as24220.n 220.158.215.20   3 u  51y   64    7  1622678  +811339   0.002`

I tried a tcpdump capture and it looks like ntpd randomly sets the "Receive Timestamp" on queries to 1/1/1970. I know receive timestamp on queries shouldn't matter, but I suspect it's not coincidence that the ntpq output reports the server hasn't been reached since 1/1/1970.

`admin@RT-AC88U-B1E8:/tmp/mnt/entware/entware/var/spool/ntp# tcpdump -i eth0 host
 ntp5.mum-in.hosts.301-moved.de -vvv
tcpdump: listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
01:48:11.323217 IP (tos 0xb8, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 76)
    x.x.x.x.ntp > ntp5.mum-in.hosts.301-moved.de.ntp: [bad udp cksum 0xe94b -> 0x6037!] NTPv4, length 48
        Client, Leap indicator: clock unsynchronized (192), Stratum 0 (unspecified), poll 6 (64s), precision -19
        Root Delay: 0.000000, Root dispersion: 0.000000, Reference-ID: (unspec)
          Reference Timestamp:  0.000000000
          Originator Timestamp: 3831673624.451940738 (2021/06/03 01:47:04)
          Receive Timestamp:    3831673624.565429405 (2021/06/03 01:47:04)
          Transmit Timestamp:   3831673691.323147763 (2021/06/03 01:48:11)
            Originator - Receive Timestamp:  +0.113488666
            Originator - Transmit Timestamp: +66.871207024
00:00:00.008176 IP (tos 0x0, ttl 52, id 63049, offset 0, flags [DF], proto UDP (17), length 76)
    ntp5.mum-in.hosts.301-moved.de.ntp > x.x.x.x.ntp: [udp sum ok] NTPv4, length 48
        Server, Leap indicator:  (0), Stratum 2 (secondary reference), poll 6 (64s), precision -23
        Root Delay: 0.022689, Root dispersion: 0.045547, Reference-ID: 14.139.60.107
          Reference Timestamp:  3831671747.264881498 (2021/06/03 01:15:47)
          Originator Timestamp: 3831673691.323147763 (2021/06/03 01:48:11)
          Receive Timestamp:    3831673691.453204151 (2021/06/03 01:48:11)
          Transmit Timestamp:   3831673691.453252667 (2021/06/03 01:48:11)
            Originator - Receive Timestamp:  +0.130056387
            Originator - Transmit Timestamp: +0.130104904
01:49:20.323109 IP (tos 0xb8, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 76)
    x.x.x.x.ntp > ntp5.mum-in.hosts.301-moved.de.ntp: [bad udp cksum 0xe94b -> 0xf5d9!] NTPv4, length 48
        Client, Leap indicator: clock unsynchronized (192), Stratum 0 (unspecified), poll 6 (64s), precision -19
        Root Delay: 0.000000, Root dispersion: 0.000000, Reference-ID: (unspec)
          Reference Timestamp:  0.000000000
          Originator Timestamp: 3831673691.453252667 (2021/06/03 01:48:11)
          Receive Timestamp:    2208988800.008178023 (1970/01/01 00:00:00)
          Transmit Timestamp:   3831673760.323004341 (2021/06/03 01:49:20)
            Originator - Receive Timestamp:  -1622684891.445074643
            Originator - Transmit Timestamp: +68.869751673
00:00:00.443431 IP (tos 0x0, ttl 52, id 7962, offset 0, flags [DF], proto UDP (17), length 76)
    ntp5.mum-in.hosts.301-moved.de.ntp > x.x.x.x.ntp: [udp sum ok] NTPv4, length 48
        Server, Leap indicator:  (0), Stratum 2 (secondary reference), poll 6 (64s), precision -23
        Root Delay: 0.022689, Root dispersion: 0.046585, Reference-ID: 14.139.60.107
          Reference Timestamp:  3831671747.264881498 (2021/06/03 01:15:47)
          Originator Timestamp: 3831673760.323004341 (2021/06/03 01:49:20)
          Receive Timestamp:    3831673760.452384601 (2021/06/03 01:49:20)
          Transmit Timestamp:   3831673760.452425126 (2021/06/03 01:49:20)
            Originator - Receive Timestamp:  +0.129380260
            Originator - Transmit Timestamp: +0.129420784
^C
4 packets captured
25 packets received by filter
0 packets dropped by kernel`

The whole time this is happening Windows 10 LAN clients cannot sync to ntpd. To rule out network configuration issues I tried running the built-in Busybox NTP and LAN clients are able to sync no problems.

I also tried running an ntp.conf with nothing except:

`pool us.pool.ntp.org`

and still experienced the same issue.

jbaker6953, 2021-06-09:
chrony also works as expected on this system.

zyxmon, 2021-06-10:
ntpd compiled in Entware with no changes from Openwrt. Probably it should be started differently.
May be this will help - https://forum.openwrt.org/t/ntp-server-does-not-work/4023

jbaker6953, 2021-06-10:
Thank you for the link. Unfortunately, that user's problem doesn't seem to be related. I don't have any problem starting ntpd manually or from init.d, it loads the appropriate config, and remains running listening on port 123. There are no other ntp processes running according to ps. It just has that strangeness posted earlier and LAN clients can't sync.

jbaker6953, 2021-06-11:
After more research I've discovered that this is a problem for many Entware applications. I thought chrony was running correctly, but it too was exhibiting the same symptoms. The issue is intermittent. While using tcpdump to study the issue I noticed that tcpdump also had the wrong time at the same time ntpd or chronyd where creating packets with the wrong receive timestamp.

So, watching several NTP transactions on my lan looks like this:

```
admin@RT-AC88U-B1E8:/tmp/etc# tcpdump -i br0 port 123 and host 192.168.1.2 -v
vv
tcpdump: listening on br0, link-type EN10MB (Ethernet), capture size 262144 byts
00:00:00.311096 IP (tos 0x0, ttl 128, id 55050, offset 0, flags [none], proto U)
    192.168.1.2.ntp > 192.168.1.1.ntp: [udp sum 8
        Client, Leap indicator: clock unsynchronized (192), Stratum 0 (unspecif3
        Root Delay: 0.055114, Root dispersion: 8.899551, Reference-ID: (unspec)
          Reference Timestamp:  3832344465.743449599 (2021/06/10 20:07:45)
          Originator Timestamp: 0.000000000
          Receive Timestamp:    0.000000000
          Transmit Timestamp:   3832373029.759450399 (2021/06/11 04:03:49)
            Originator - Receive Timestamp:  0.000000000
            Originator - Transmit Timestamp: 3832373029.759450399 (2021/06/11 0)
04:03:49.107179 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto UDP (17))
    192.168.1.1.ntp > 192.168.1.2.ntp: [bad udp 8
        Server, Leap indicator:  (0), Stratum 3 (secondary reference), poll 10 2
        Root Delay: 0.051300, Root dispersion: 0.004180, Reference-ID: voipmonim
          Reference Timestamp:  3832371061.598493037 (2021/06/11 03:31:01)
          Originator Timestamp: 3832373029.759450399 (2021/06/11 04:03:49)
          Receive Timestamp:    2208989448.703732420 (1970/01/01 00:10:48)
          Transmit Timestamp:   3832373029.109460808 (2021/06/11 04:03:49)
            Originator - Receive Timestamp:  -1623383581.055717978
            Originator - Transmit Timestamp: -0.649989590
04:05:15.828307 IP (tos 0x0, ttl 128, id 55054, offset 0, flags [none], proto U)
    192.168.1.2.ntp > 192.168.1.1.ntp: [udp sum 8
        Client, Leap indicator: clock unsynchronized (192), Stratum 0 (unspecif3
        Root Delay: 0.055114, Root dispersion: 8.899551, Reference-ID: (unspec)
          Reference Timestamp:  3832344465.732624899 (2021/06/10 20:07:45)
          Originator Timestamp: 0.000000000
          Receive Timestamp:    0.000000000
          Transmit Timestamp:   3832373116.482626199 (2021/06/11 04:05:16)
            Originator - Receive Timestamp:  0.000000000
            Originator - Transmit Timestamp: 3832373116.482626199 (2021/06/11 0)
04:05:15.828824 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto UDP (17))
    192.168.1.1.ntp > 192.168.1.2.ntp: [bad udp 8
        Server, Leap indicator:  (0), Stratum 3 (secondary reference), poll 10 2
        Root Delay: 0.051300, Root dispersion: 0.004272, Reference-ID: voipmonim
          Reference Timestamp:  3832371061.598493037 (2021/06/11 03:31:01)
          Originator Timestamp: 3832373116.482626199 (2021/06/11 04:05:16)
          Receive Timestamp:    3832373115.830616489 (2021/06/11 04:05:15)
          Transmit Timestamp:   3832373115.831061522 (2021/06/11 04:05:15)
            Originator - Receive Timestamp:  -0.652009710
            Originator - Transmit Timestamp: -0.651564677
00:00:00.822257 IP (tos 0x0, ttl 128, id 55056, offset 0, flags [none], proto U)
    192.168.1.2.ntp > 192.168.1.1.ntp: [udp sum 8
        Client, Leap indicator: clock unsynchronized (192), Stratum 0 (unspecif3
        Root Delay: 0.053726, Root dispersion: 9.417495, Reference-ID: (unspec)
          Reference Timestamp:  3832373116.508524299 (2021/06/11 04:05:16)
          Originator Timestamp: 0.000000000
          Receive Timestamp:    0.000000000
          Transmit Timestamp:   3832373152.914525099 (2021/06/11 04:05:52)
            Originator - Receive Timestamp:  0.000000000
            Originator - Transmit Timestamp: 3832373152.914525099 (2021/06/11 0)
04:05:52.261366 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto UDP (17))
    192.168.1.1.ntp > 192.168.1.2.ntp: [bad udp 8
        Server, Leap indicator:  (0), Stratum 3 (secondary reference), poll 9 (2
        Root Delay: 0.051300, Root dispersion: 0.004318, Reference-ID: voipmonim
          Reference Timestamp:  3832371061.598493037 (2021/06/11 03:31:01)
          Originator Timestamp: 3832373152.914525099 (2021/06/11 04:05:52)
          Receive Timestamp:    2208989449.214893200 (1970/01/01 00:10:49)
          Transmit Timestamp:   3832373152.263588458 (2021/06/11 04:05:52)
            Originator - Receive Timestamp:  -1623383703.699631899
            Originator - Transmit Timestamp: -0.650936641
00:00:00.806610 IP (tos 0x0, ttl 128, id 55058, offset 0, flags [none], proto U)
    192.168.1.2.ntp > 192.168.1.1.ntp: [udp sum 8
[... 116 lines cut ...]
            Originator - Receive Timestamp:  -1623383838.864764514
            Originator - Transmit Timestamp: -0.652597627
04:08:25.609907 IP (tos 0x0, ttl 128, id 55072, offset 0, flags [none], proto U)
    192.168.1.2.ntp > 192.168.1.1.ntp: [udp sum 8
        Client, Leap indicator: clock unsynchronized (192), Stratum 0 (unspecif3
        Root Delay: 0.055053, Root dispersion: 9.418533, Reference-ID: (unspec)
          Reference Timestamp:  3832373245.514594399 (2021/06/11 04:07:25)
          Originator Timestamp: 0.000000000
          Receive Timestamp:    0.000000000
[... cut ...]
          Originator Timestamp: 3832373306.264595099 (2021/06/11 04:08:26)
          Receive Timestamp:    3832373305.612141027 (2021/06/11 04:08:25)
          Transmit Timestamp:   3832373305.612588678 (2021/06/11 04:08:25)
            Originator - Receive Timestamp:  -0.652454072
            Originator - Transmit Timestamp: -0.652006421
```

Note that every time a transaction is successful the timestamp from tcpdump is accurate, but every time the transaction yields a response with a bad received timestamp that tcpdump's timestamp is also 00:00:00.xxxxxx.

jbaker6953, 2021-06-12:
I have narrowed this down to either a bug in Broadcom's cut-through forwarding, or some interaction between Entware and CTF. When CTF is enabled on the router there are numerous packets with bogus timestamps. When CTF is disabled the bogus timestamps go away. For some reason built-in utilities in the firmware aren't affected, but Entware utilities that use network timestamps are crippled on my device when CTF is enabled.

zyxmon, 2021-06-12:
I am sure it is CTF related issue. CTF is some sort of hardware acceleration (Cut Through Forwarding). Firmware knows about it and and works correctly. Entware is a general purpose repository - some network utilities mat not work for hardware accelerated devices.
This happened earlier with Mediatec routers with HWNAT.

closed 2021-06-12 by zyxmon