Entware/Entware #664: Why destroy /opt ?

cwmoriarty, 2021-06-10:
I use `/opt/containerd` and would like to not have to move it to `/volume1/@Entware/opt` after installation

More info:
https://github.com/containerd/containerd/commit/1537f31381b946f395348d9891602a1e4a2c1070

zyxmon, 2021-06-10:
I guess you are using Entware on Qnap. In the time Entware qpkg was created there were no rules for folders that packages can use. Many packages used /Apps folder. No packages used /opt. Entware on Qnap. Entware was made as a replacement for Optware and it removes /opt when uninstalled,
Qnap should develop rules for packages and folder usage and qpkg developers will follow it.

closed 2021-06-10 by zyxmon

cwmoriarty, 2021-06-10:
Understood, and FWIW this is running on Synology DSM7.

zyxmon, 2021-06-10:
We do not have any special binaries or scripts for Synology. May be some third party script, binary or instruction destroys /opt. In case you think wiki is not correct you can edit it and fix.

zocoloco, 2022-09-28:
@cwmoriarty did you ever get this resolved? I'm in a similar situation. I'm installing Homebridge on my Synology NAS DSM 7. In the Entware installaiton instructions it says to do the following:
`rm -rf /opt
mkdir /opt
mount -o bind "/volume1/@Entware/opt" /opt 
 `
This would remove anything that's installed in /opt. I too have containered installed (in opt)

bao3, 2024-05-31:
the same issue .  In Synology  DSM 7 , there is a folder named "containerd" in the `/opt`, so I cannot do ` rm -rf /opt` .
How could Synology DSM 7 user install Entware safely ?

ryzhovau, 2024-06-01:
There's no any graceful solution, until Entware and DSM both uses `/opt`.

adamlogan, 2024-12-24:
Backed up /opt/containerd and proceeded to try to install Entware anyway. The directories were empty, so I figured it probably wouldn't be a big deal. Big mistake. Now I can't log into Synology DSM, I keep getting error "You are not authorized to use this service." *sigh* 

Tried uninstalling, and restored /opt/containerd rebooted, still can't access Synology DSM. If anyone has pointers on how to fix this, please let me know.

asdfgeoff, 2025-01-01:
> The directories were empty, so I figured it probably wouldn't be a big deal. Big mistake.

@adamlogan appreciate you sharing this, as you have dissuaded me from attempting the same thing myself.

> Tried uninstalling, and restored /opt/containerd rebooted, still can't access Synology DSM.

Shot-in-the-dark, but: have you confirmed that the permissions of the restored files are identical as before?

Looking at my own NAS, both `/opt` and its sub-folders (`/opt/bin` and `/opt/lib`) are `drwx--x--x` and owned by `root` / `root`.

koszta5, 2025-03-14:
@adamlogan  at least you just saved me from doing the same mistake

adamlogan, 2025-12-14:
I was able to recover my NAS after several days of poking around as a CLI noob. I vaguely remember that somehow the Entware install process screwed with my user's password for the diskstation webapp. If I recall correctly, I was still able to get in as the admin user via ssh. After soft resetting the NAS, I was able to SSH in and fix syntax for /etc/passwd string for my primary user. I think I had to manually create a new password via the synouser command. See `sudo synouser --help` for the syntax.

If you're after Linux utilities on the NAS, you're better of using the stuff available in the community repo, search for "SynoCLI" and several packages will pop up with collections of CLI utilities will show up that will work on the NAS. Or better still, get a docker container running on Container Manager and go with a pre-built container image or build your own with the cli tools or applications you want.

I have not tried to install Entware again since that debacle.