Entware/Entware #665: Building Entware Tools Fails Dependancy python 2.7

garycnew, 2021-06-20:
# cd Entware/
# make tools/install
/tmp/home/root/Entware/scripts/config.guess: line 1: command: not found
/tmp/home/root/Entware/scripts/config.guess: line 1: command: not found
/tmp/home/root/Entware/scripts/config.guess: line 1: command: not found
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'install'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python2'... failed.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'rsync'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.

Build dependency: Please install Python 2.7

Prerequisite check failed. Use FORCE=1 to override.
make[2]: *** [/tmp/home/root/Entware/include/toplevel.mk:180: staging_dir/host/.prereq-build] Error 1
make[1]: *** [/tmp/home/root/Entware/include/toplevel.mk:77: prepare-tmpinfo] Error 2
make: *** [/tmp/home/root/Entware/include/toplevel.mk:228: tools/install] Error 2

I have all the build dependancies satisfied except for python 2.7, which is no longer available in the current repository. 

# opkg update 
Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz 
Updated list of available packages in /opt/var/opkg-lists/entware 

I attempted to install the archived version of python 2.7, but opkg complains with the following: 

# curl --output python_2.7.17-2_armv7-2.6.ipk http://bin.entware.net/armv7sf-k2.6/archive/python_2.7.17-2_armv7-2.6.ipk 

# opkg install ./python_2.7.17-2_armv7-2.6.ipk 
Unknown package 'python'. 
Collected errors: 
* pkg_hash_fetch_best_installation_candidate: Packages for python found, but incompatible with the architectures configured 
* opkg_install_cmd: Cannot install package python. 

# cat /opt/etc/entware_release
release=entware
arch=arm
cpu=cortex-a9
cpu_subtype=unknown
float=soft
gcc=8.4.0

# opkg print-architecture
arch all 100
arch armv7-2.6 160

Python3 & dependancies install without issue: 

# opkg info python3
Package: python3
Version: 3.9.4-1
Depends: libc, libssp, librt, libpthread, python3-light, python3-unittest, python3-ncurses, python3-ctypes, python3-pydoc, python3-logging, python3-decimal, python3-multiprocessing, python3-codecs, python3-gdbm, python3-sqlite3, python3-xml, python3-distutils, python3-email, python3-openssl, python3-urllib, python3-cgi, python3-cgitb, python3-dbm, python3-lzma, python3-asyncio
Status: install user installed
Section: lang
Architecture: armv7-2.6
Size: 971
Filename: python3_3.9.4-1_armv7-2.6.ipk
Description: This package contains the (almost) full Python install.
It's python3-light + all other packages.
Installed-Time: 1623987187

I find it ironic that Entware no longer provides the required python 2.7 package(s) that is required to build Entware tools. 

Your guidance is greatly appreciated. 

Respectfully, 

Gary

zyxmon, 2021-06-20:
We do not provide this sort of support. Python 2.7 is still in Debian. We recommend Debian to crosscompile packages.

closed 2021-06-20 by zyxmon

garycnew, 2021-06-20:
L@m3

yoyoma2, 2024-06-11:
Shouldn't this issue be re-opened? This error happens when crosscompiling packages on ubuntu 23.10 which doesn't allow installing python 2.7. It seems [debian also removed it](https://wiki.debian.org/Python).

The workaround of commenting out the python 2.7 prerequisite given above by @garycnew at least allows building specific packages (most don't need python 2.7).

ryzhovau, 2024-06-11:
k2.6-based feed are EoL. 
https://entware.net/2024/02/23/Changelog.html

yoyoma2, 2024-06-11:
I was using config file **armv7-3.2.config** copied to .config and crosscompiling on **ubuntu 23.10** and the build failed with the error in this issue. The @garycnew workaround let the build progress more but eventually failed due to no python 2.7. I was still able to build the few packages I needed thanks to @garycnew's workaround.

The python 2.7 dependency build error isn't specific to the k2.6 feed nor to the native compilation.

ryzhovau, 2024-06-12:
No I get it:)

Please, use Docker environment or install necessary deps from our [Dockerfile](https://github.com/Entware/docker/blob/main/Dockerfile).

garycnew, 2024-06-12:
It's nice to see I'm not the only one with this issue and that my workaround comment is helpful.

I'm curious when the Entware MakeFile might be updated to exclude python-2.7?

Thanks.

yoyoma2, 2024-06-13:
The docker way of building should be promoted [on the wiki](https://github.com/Entware/Entware/wiki/Compile-packages-from-sources). I let it build out of curiosity and it failed after a long time while building wstunnel which is trying to access https://github.com/erebe/fast-socks5 which seems not to exist.

The docker method does resolve the python 2.7 issue and the wstunnel error is a different issue so I'm done here.

ryzhovau, 2024-06-13:
Wiki is available for editing for everyone:)

yoyoma2, 2024-06-13:
Added a reference to building with docker [here](https://github.com/Entware/Entware/wiki/Compile-packages-from-sources/_compare/23d709a1661a09270a66d648f7bbda3a08e829f6...565bf98c22e02cec66e7d7eac29c49937bb9dd86).