Entware/Entware #667: Please update miniupnpd to the latest version and compile it with IPv6 support

arbv, 2021-06-21:
As IPv6 deployment getting more traction these days, it might be a good idea to update `miniupnpd` to include IPv6 support. Doing so requires adding `ip6tables` as well.

Most of the routers do provide `miniupnpd` as part of their firmware. Unfortunately, even in the routers which are nominally compatible with IPv6 often the `miniupnpd` daemon is compiled without support for it, leaving the device without any Port Control Protocol implementation and, thus, a way to punch holes in its firewall for applications in the LAN which might require it (p2p file sharing software, video conferencing software, etc). Having the `miniupnpd` compiled with IPv6 protocol in Entware is a way to solve the issue. Alternatively, `miniupnpd` with IPv6 support might be compiled as a separate package (e.g. `miniupnpd-ipv6`).

ryzhovau, 2021-07-06:
Package updated https://github.com/Entware/entware-packages/commit/530e3862f013a576c4dcad4f4a3c0cdeb0b5f478
Binaries will arrive next week. We'll keep it in current state to simplify sync with OpenWrt.

closed 2021-07-06 by ryzhovau

arbv, 2021-07-08:
Thanks! That is super cool!