Entware/Entware #669: Support for arm64 toolchain?

djeikyb, 2021-07-14:
I've got an m1 / apple silicon macbook air and looking to build the toolchain. I started out experimenting in a debian docker image. So far my only trouble was go-bootstrap. I got a 404 before modifying the makefile. I made this change to use 1.16.3 instead of 1.4.3, and to request the arm64 bintype instead of aarch64:

```
diff --git a/tools/go-bootstrap/Makefile b/tools/go-bootstrap/Makefile
index 8a00eebb..40888138 100644
--- a/tools/go-bootstrap/Makefile
+++ b/tools/go-bootstrap/Makefile
@@ -7,15 +7,16 @@
 include $(TOPDIR)/rules.mk

 PKG_NAME:=go-bootstrap
-PKG_VERSION:=1.4.3
+PKG_VERSION:=1.16.3
 OS=$(shell uname | tr 'A-Z' 'a-z')
 bintype:=$(shell uname -m | sed \
         -e 's/i[3-9]86/386/' \
         -e 's/x86_64/amd64/' \
+        -e 's/aarch64/arm64/' \
 )

 PKG_SOURCE:=go$(PKG_VERSION).$(OS)-$(bintype).tar.gz
-PKG_HASH:=ce3140662f45356eb78bc16a88fc7cfb29fb00e18d7c632608245b789b2086d2
+PKG_HASH:=566b1d6f17d2bc4ad5f81486f0df44f3088c3ed47a3bec4099d8ed9939e90d5d
 PKG_SOURCE_URL:=https://storage.googleapis.com/golang
 HOST_BUILD_DIR:=$(STAGING_DIR_HOST)/go1.4
 HOST_UNPACK:=gzip -dc $(DL_DIR)/$(PKG_SOURCE) | tar -C $(HOST_BUILD_DIR) --strip-components=1 -xf -
```

1. Is the sed expression addition appropriate to pr? Should it be other places too?
2. This isn't my bailiwick, but skimming the golang issues, it sounds like bootstrapping go with version 1.4.3 isn't an option for arm64. But I'm not entirely sure what version ought to be used?

ryzhovau, 2021-07-14:
Looks like all Entware maintainers are using Debian stable x64 and didn't noticed about this issue. 
I'm not sure it's possible to build all existent packages on arm64, some of them requires tricky host-compilation before further cross-compilation. Anyway, thanks for report!

djeikyb, 2021-07-19:
Oops dunno how I closed this!

ryzhovau, 2021-07-20:
Hey, @djeikyb. Did you tried to compile some packages on m1/apple host? Still not sure `-e 's/aarch64/arm64/'` should be applied.

djeikyb, 2021-07-20:
Yeah, I actually got pretty far I think? I had to take a break for the work week, but I plan on coming back to this and reporting my hacky diffs. I started out with `make -j7` cause why not. But then something would fail. Before trying anything else I'd try `make -j2` or `make -j1`. Sometimes it just seems like a race condition between dependencies? But other times I'd find out I needed to install some package. Then back to `make -j7` for the speed.. so I tell myself.. maybe should stick to 1 or 2 lol. The golang package was the first I had to actually resolve a problem.

zyxmon, 2021-07-20:
>> Sometimes it just seems like a race condition between dependencies

Probably. In case `make -j7` fails on a package xxx I run `make package/xxx/compile` and if it is ok I run `make -j7`. The developers have different hardware with different number of cores (-jN). The current trunk was built with no race conditions for 4 different architectures in my case.

djeikyb, 2021-07-21:
Also, in my debian buster:10 container, I've collected these required dependencies:

```
apt-get install -y bzip2 cmake curl g++ gawk git gzip gzip2 libncurses-dev libpcre2-dev libssl-dev make nodejs python2 python3 rsync unzip wget zlib1g-dev
```

I rather belatedly saw the secondary readme which generically mentions some of these packages.

zyxmon, 2021-07-21:
The current wiki does not contain host packages list and refers to openwrt documentation.
In 2015 the packages list was something like
```
$ sudo apt-get install autoconf automake bash bison bzip2 cvs diffutils file flex g++ \
gawk gettext git-core gperf groff-base libexpat1-dev libncurses-dev libssl-dev libtool \
libslang2 libxml-parser-perl make patch perl python ruby sed shtool subversion tar \
texinfo unzip zlib1g zlib1g-dev pkg-config gettext libgmp3-dev libmpfr-dev libmpc-dev
```

closed 2021-08-20 by no-response[bot]

Anonym-tsk, 2024-09-04:
```
+ curl -f --connect-timeout 20 --retry 5 --location https://storage.googleapis.com/golang/go1.20.6.linux-aarch64.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/go1.20.6.linux-aarch64.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 

```