Entware/Entware #670: php8

kimdoyeonsa, 2021-07-22:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

https://downloads.openwrt.org/releases/packages-21.02/arm_cortex-a9/packages/php8-cgi_8.0.3-1_arm_cortex-a9.ipk
.
.
.
https://downloads.openwrt.org/releases/packages-21.02/arm_cortex-a9/packages/php8_8.0.3-1_arm_cortex-a9.ipk
https://www.php.net/

php8 all Packages

I want to install php 8.0. 

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.
aarch64-k3.10
armv5sf-k3.2 
armv7sf-k2.6
armv7sf-k3.2
mipselsf-k3.4
mipssf-k3.4
x64-k3.2
x86-k2.6

butaford, 2021-07-23:
Without official support from Entware developers:
https://fw.pnpz.xyz/entware/repo-php8

kimdoyeonsa, 2021-07-24:
repo-php8

PHP Startup: Unable to load dynamic library 'intl.so' (tried: /opt/lib/php/intl.so (libicuio.so.68: cannot open shared object file: No such file or directory), /opt/lib/php/intl.so.so (/opt/lib/php/intl.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0

butaford, 2021-07-24:
add repo in /opt/etc/opkg.conf: 
src/gz fwpnpz_php8 http://pnpz.xyz/files/fw/entware/repo-php8/_aarch64-3.10_
opkg update
opkg install php8-cli

ldd /opt/bin/php8-cli
/opt/bin/php8-cli -v
[image]

The-BB, 2021-07-24:
check the configuration file /opt/etc/php.ini
https://github.com/Entware/entware-packages/blob/master/lang/php8/files/php.ini#L128

kimdoyeonsa, 2021-07-24:
opkg install failed
wget: not an http or ftp url: https://fw.pnpz.xyz/entware/repo-php8/armv7sf-k3.2/Packages.gz
*** Failed to download the package list from https://fw.pnpz.xyz/entware/repo-php8/armv7sf-k3.2/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://fw.pnpz.xyz/entware/repo-php8/armv7sf-k3.2/Packages.gz, wget returned 1.

butaford, 2021-07-24:
`opkg install ca-bundle`

kimdoyeonsa, 2021-07-24:
[](url) 
[image]

butaford, 2021-07-24:
**more attentively:**
 https://fw.pnpz.xyz/entware/repo-php8/**armv7-3.2**

kimdoyeonsa, 2021-07-25:
php8-mod-intl
libicuio.so.68
libicuuc.so.68
libicudata.so.68
libicui18n.so.68
library file not found

kimdoyeonsa, 2021-07-29:
php 8-mod-intl 
[
[image]
](url)

[
[image]
](url)

ryzhovau, 2021-08-20:
Done.

closed 2021-08-20 by ryzhovau