Entware/Entware #674: libssl-dev (for Perl applications like FHEM or Webmin)

BerlinSnoop, 2021-08-14:
Some perl modules like 'Net::SSLeay' could not be compiled with gcc without it...

SSLeay.xs:163:10: fatal error: openssl/err.h: No such file or directory
 #include <openssl/err.h>

- [ x] No similar package already added and/or this one got some new killer feature(s),
- [ x] Package is in active development (at least, not abandoned for 5 years or more),
- [ x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

zyxmon, 2021-08-15:
There no dev packages in Entware (and in Openwrt) with few exceptions. Please follow wiki https://github.com/Entware/Entware/wiki/Self-installation-of-perl-modules
This will install missing headers - https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation

closed 2021-08-15 by zyxmon

BerlinSnoop, 2021-08-19:
Many Thanks. I have followed your advice und it has worked 👍 on my Synology DS418 (Arm64). Now I can compile many more Perl modules for FHEM and Webmin can update itself (SSL stuff was missing).