Entware/Entware #678: php8-mod-gmp : Error relocating libgmp.so.10: obstack_vprintf: symbol not found

vonguyenkhang, 2021-08-28:
**Describe the bug**
php8-mod-gmp : Error relocating libgmp.so.10: obstack_vprintf: symbol not found

**Environment (please complete the following information):**
 - Entware feed: [armv7sf-k3.2](http://bin.entware.net/armv7sf-k3.2/Packages.gz)
 - Device: [D-Link DIR-868L Rev.A1](https://wiki.dd-wrt.com/wiki/index.php/D-Link_DIR-868L)
 - Firmware version: [DD-WRT v3.0-r47256 std (08/25/21)](https://forum.dd-wrt.com/phpBB2/viewtopic.php?p=1244289#1244289)

```sh
root@RB:~# php -v

Warning: PHP Startup: Unable to load dynamic library '/opt/lib/php8/gmp.so' (tried: /opt/lib/php8/gmp.so (Error relocating /opt/lib/libgmp.so.10: obstack_vprintf: symbol not found), /etc/php/modules//opt/lib/php8/gmp.so.so (Error loading shared library /etc/php/modules//opt/lib/php8/gmp.so.so: No such file or directory)) in Unknown on line 0
PHP 8.0.6 (cli) (built: May 11 2021 15:56:16) ( NTS )
Copyright (c) The PHP Group
Zend Engine v4.0.6, Copyright (c) Zend Technologies
    with Zend OPcache v8.0.6, Copyright (c), by Zend Technologies
```

The-BB, 2021-08-28:
`opkg install php8 --force-overwrite` or if you want to save php.ini check the path to the modules in it, the path should be `/opt/lib/php8`
```
~ # cat /opt/etc/php.ini | grep /lib
extension_dir = "/opt/lib/php8"
~ # 
```

zyxmon, 2021-08-28:
No problems with clean installation. Closing.
```
/share/CACHEDEV1_DATA/.qpkg/Entware/root # opkg install php8-mod-gmp
Installing php8-mod-gmp (8.0.9-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/php8-mod-gmp_8.0.9-1_armv7-3.2.ipk
Installing libpcre2 (10.37-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libpcre2_10.37-1_armv7-3.2.ipk
Installing libxml2 (2.9.12-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libxml2_2.9.12-1_armv7-3.2.ipk
Installing zoneinfo-core (2021a-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/zoneinfo-core_2021a-1_armv7-3.2.ipk
Installing php8 (8.0.9-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/php8_8.0.9-1_armv7-3.2.ipk
Installing libgmp (6.2.1-1a) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libgmp_6.2.1-1a_armv7-3.2.ipk
Configuring libxml2.
Configuring libpcre2.
Configuring zoneinfo-core.
Configuring php8.
Configuring libgmp.
Configuring php8-mod-gmp.
/share/CACHEDEV1_DATA/.qpkg/Entware/root # opkg install php8-cli
Installing php8-cli (8.0.9-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/php8-cli_8.0.9-1_armv7-3.2.ipk
Configuring php8-cli.
/share/CACHEDEV1_DATA/.qpkg/Entware/root # php-cli -v
PHP 8.0.9 (cli) ( NTS )
Copyright (c) The PHP Group
Zend Engine v4.0.9, Copyright (c) Zend Technologies
```

closed 2021-08-28 by zyxmon

vonguyenkhang, 2021-08-29:
Everything are good after switching from DDWRT's bundled php8 to Entware's php8, I'll report this issue to DDWRT forum.
Sorry for the false report. And thanks for the super fast response.
Stay safe.