Entware/Entware #679: named-compilezone is missing in BIND9 related packages

arbv, 2021-09-03:
The `named-compilezone` tool, which is intended to be used to convert zone files between different formats, is missing in BIND9 related packages. Could it be included into `bind-check` package, alongside `named-checkzone`?

The-BB, 2021-09-25:
You can create it manually if you want.
```
~ # cat bind-9.17.13/bin/check/Makefile | grep -C 2 'compilezone'
install-exec-hook:
	ln -f $(DESTDIR)$(bindir)/named-checkzone \
	      $(DESTDIR)$(bindir)/named-compilezone
...
~ # 
```

arbv, 2021-11-14:
Thanks! Still, I think it would be better to make it available after package installation, especially considering that it does not take any space.

ryzhovau, 2021-12-09:
Bind came from OpenWrt with no changes. We'll keep it in this state to simplify next sync.

closed 2021-12-09 by ryzhovau