Entware/Entware #681: Segmentation fault in urbackup-server 2.4.13-1

barnumbirr, 2021-09-08:
`urbackup-server 2.4.13-1` segfaults after rebooting the device for the first time after installation. Here's the only bit of information I get:

```bash
[~] # urbackupsrv run -u admin
2021-09-08 12:23:42: Starting HTTP-Server on port 55414
2021-09-08 12:23:42: HTTP: Server started up successfully!
```

**Environment:**
 - Entware feed: x64-3.2
 - Device: QNAP TS-873A
 - Firmware version: QuTS Hero 4.5.3.1698

barnumbirr, 2021-09-08:
About 1 out of 100 times, `urbackupsrv` manages to start a bit further.

```bash
[~] # urbackupsrv run -u admin
2021-09-09 01:23:30: Starting HTTP-Server on port 55414
2021-09-09 01:23:30: HTTP: Server started up successfully!
2021-09-09 01:23:30: SQLite: recovered 1178 frames from WAL file /share/ZFS530_DATA/.qpkg/Entware/var/urbackup/backup_server.db-wal code: 283
2021-09-09 01:23:30: SQLite: recovered 96 frames from WAL file /share/ZFS530_DATA/.qpkg/Entware/var/urbackup/backup_server_files.db-wal code: 283
2021-09-09 01:23:30: SQLite: recovered 6404 frames from WAL file /share/ZFS530_DATA/.qpkg/Entware/var/urbackup/backup_server_link_journal.db-wal code: 283
2021-09-09 01:23:31: SQLite: recovered 13370 frames from WAL file /share/ZFS530_DATA/.qpkg/Entware/var/urbackup/backup_server_links.db-wal code: 283
2021-09-09 01:23:31: SQLite: recovered 3 frames from WAL file /share/ZFS530_DATA/.qpkg/Entware/var/urbackup/backup_server_settings.db-wal code: 283
Segmentation fault
```

ryzhovau, 2021-09-09:
Can't reproduce on MIPS router, works for me.

barnumbirr, 2021-09-09:
How would I go about compiling the package with debug symbols enabled?
I've tried adding `PKG_CONFIG_DEPENDS := CONFIG_DEBUG` when cross-compiling but gdb still complains:

```bash
Reading symbols from urbackupsrv...
(No debugging symbols found in urbackupsrv)
```

zyxmon, 2021-09-10:
I checked if urbackup runs on my Qnap x64 NAS. No segfaults, web interface works on port 55414.
```
/share/CACHEDEV1_DATA/.qpkg/Entware/root # /opt/etc/init.d/S60urbackup-server start
 Starting urbackupsrv...              done.
/share/CACHEDEV1_DATA/.qpkg/Entware/root # /opt/etc/init.d/S60urbackup-server check
 Checking urbackupsrv...              alive.
/share/CACHEDEV1_DATA/.qpkg/Entware/root # cat /etc/model.conf
[System Enclosure]
VENDOR = QNAP
MODEL = TS-253B
......
```
I have added `-u root` to startup script. I have root user on my qnap (because of alternative installation script)

barnumbirr, 2021-09-10:
Not sure if that makes a difference but `urbackup-server` ran fine, even after reboots before any client backup data was added. First reboot after running backups on 5 clients makes it segfault.

zyxmon, 2021-09-10:
I have tried urbackup server on x64 Qnap NAS with one client and file backups. Stopped and started server many times manually with no problems.
Try to disable server autostart and run it manually. May be it starts too eraly after reboot.

barnumbirr, 2021-09-10:
> Try to disable server autostart and run it manually. May be it starts too early after reboot.

Tried that already unfortunately. It's not just running urbackup that fails, all commands segfault:

```bash
[admin@QNAP ~]# urbackupsrv export-auth-log
Segmentation fault
```

barnumbirr, 2021-09-15:
Issue still persists, haven't figured out how to build "debug" package yet.

barnumbirr, 2021-09-15:
Started from scratch:
- reset NAS to factory settings
- recreated "System" Storage pool
- reinstalled Entware 1.03std
- reinstalled and reconfigure urbackupserver
- ran full image backup for one host (about 200GiB of data)
- rebooted NAS
Same outcome.

barnumbirr, 2021-09-17:
Anyone to give me a few pointers as to how to compile `urbackup-server` with debug symbols enabled? Have been struggling with this for over a week now. Knowing what actually happens when `urbackup-server` segfaults will be uselful.

barnumbirr, 2021-09-18:
@ryzhovau @zyxmon Any ideas?

zyxmon, 2021-09-19:
@barnumbirr we do not support package compilation. Compilation flags can be found in package Makefile and rules.mk aтв .config.

zyxmon, 2021-09-19:
@barnumbirr - have you generated certificates in /opt/var/cache/pixelserv directory. No crashes if there are certificates where expected.

barnumbirr, 2021-09-19:
@zyxmon `TARGET_CFLAGS += -ggdb3` doesn't work though. Are there any other flags I need to set in the Makefile?

barnumbirr, 2021-09-23:
Still not able to compile the package with debug symbols enabled so still no way for me to figure out what's wrong.
Anyone have any pointers as to what I'm doing wrong?

barnumbirr, 2021-09-28:
Closing this as required dbgsym package cannot be compiled and no further information can be provided with it.

barnumbirr, 2022-01-30:
Finally managed to get `gdb` and `gdbserver` to play nice, here's the output:

```
$ ./scripts/remote-gdb 10.0.1.100:9999 ./build_dir/target-x86_64_glibc-2.27/urbackup-server-2.4.13/urbackupsrv
Using target x86_64 (glibc, -2.27)
GNU gdb (GDB) 11.1
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=x86_64-openwrt-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build_dir/target-x86_64_glibc-2.27/urbackup-server-2.4.13/urbackupsrv...
0x00007ffff7fd8090 in _start ()
   from /root/Entware/scripts/../staging_dir/target-x86_64_glibc-2.27/root-x64-3.2/opt/lib/ld-linux-x86-64.so.2
(gdb) set remote exec-file /opt/bin/urbackupsrv
(gdb) set args run -u admin -v debug
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /root/Entware/build_dir/target-x86_64_glibc-2.27/urbackup-server-2.4.13/urbackupsrv run -u admin -v debug
[New Thread 9407.9409]
[New Thread 9407.9410]

Thread 1 "urbackupsrv" received signal SIGSEGV, Segmentation fault.
0x000000000055dece in sqlite3WalFindFrame (pWal=0xbdb948, pgno=1, piRead=piRead@entry=0x7fffffffb394) at sqlite/sqlite3.c:60697
60697       for(iKey=walHash(pgno); sLoc.aHash[iKey]; iKey=walNextHash(iKey)){
(gdb) bt
#0  0x000000000055dece in sqlite3WalFindFrame (pWal=0xbdb948, pgno=1, piRead=piRead@entry=0x7fffffffb394)
    at sqlite/sqlite3.c:60697
#1  0x000000000055e002 in readDbPage (pPg=pPg@entry=0xbf0f88) at sqlite/sqlite3.c:53112
#2  0x000000000056608f in getPageNormal (pPager=0xbbc758, pgno=1, ppPage=0x7fffffffb400, flags=<optimized out>)
    at sqlite/sqlite3.c:55607
#3  0x0000000000525fdd in sqlite3PagerGet (flags=0, ppPage=0x7fffffffb400, pgno=1, pPager=<optimized out>)
    at sqlite/sqlite3.c:55722
#4  btreeGetPage (pBt=pBt@entry=0xbba518, pgno=pgno@entry=1, ppPage=ppPage@entry=0x7fffffffb460, flags=flags@entry=0)
    at sqlite/sqlite3.c:64699
#5  0x0000000000567291 in lockBtree (pBt=0xbba518) at sqlite/sqlite3.c:65657
#6  sqlite3BtreeBeginTrans (p=0xbba448, wrflag=wrflag@entry=0, pSchemaVersion=pSchemaVersion@entry=0x0)
    at sqlite/sqlite3.c:493
#7  0x0000000000595211 in sqlite3InitOne (db=0xbbbe08, iDb=iDb@entry=0, pzErrMsg=pzErrMsg@entry=0x7fffffffc248,
    mFlags=mFlags@entry=0) at sqlite/sqlite3.c:122405
#8  0x00000000005953cc in sqlite3Init (db=db@entry=0xbbbe08, pzErrMsg=pzErrMsg@entry=0x7fffffffc248)
    at sqlite/sqlite3.c:122590
#9  0x000000000059540f in sqlite3ReadSchema (pParse=pParse@entry=0x7fffffffc240) at sqlite/sqlite3.c:122616
#10 0x00000000005a1a50 in sqlite3Pragma (pParse=pParse@entry=0x7fffffffc240, pId1=pId1@entry=0x7fffffffb888,
    pId2=pId2@entry=0x7fffffffb8a0, pValue=pValue@entry=0x7fffffffb8d0, minusFlag=minusFlag@entry=0)
    at sqlite/sqlite3.c:120139
#11 0x00000000005a4567 in yy_reduce (yyLookahead=<optimized out>, pParse=0x7fffffffc240, yyruleno=232,
    yypParser=0x7fffffffb840, yyLookaheadToken=...) at sqlite/sqlite3.c:149648
#12 sqlite3Parser (yyminor=..., yymajor=<optimized out>, yyp=0x7fffffffb840) at sqlite/sqlite3.c:19102
#13 sqlite3RunParser (pParse=0x7fffffffc240, zSql=0xbdb371 "", pzErrMsg=0x7fffffffc238) at sqlite/sqlite3.c:20225
#14 0x00000000005a97d4 in sqlite3Prepare (db=db@entry=0xbbbe08, zSql=zSql@entry=0xbbc308 "PRAGMA synchronous=NORMAL",
    nBytes=nBytes@entry=25, prepFlags=prepFlags@entry=128, pReprepare=pReprepare@entry=0x0,
    ppStmt=ppStmt@entry=0x7fffffffc4e8, pzTail=0x7fffffffc4f0) at sqlite/sqlite3.c:122806
#15 0x00000000005a9c0f in sqlite3LockAndPrepare (db=0xbbbe08, zSql=0xbbc308 "PRAGMA synchronous=NORMAL", nBytes=25,
    prepFlags=prepFlags@entry=128, pOld=pOld@entry=0x0, ppStmt=ppStmt@entry=0x7fffffffc4e8, pzTail=0x7fffffffc4f0)
    at sqlite/sqlite3.c:122906
--Type <RET> for more, q to quit, c to continue without paging--
#16 0x00000000005a9f46 in sqlite3_prepare_v2 (db=<optimized out>, zSql=<optimized out>, nBytes=<optimized out>,
    ppStmt=ppStmt@entry=0x7fffffffc4e8, pzTail=pzTail@entry=0x7fffffffc4f0) at sqlite/sqlite3.c:122989
#17 0x00000000004c8c76 in CDatabase::Prepare (this=this@entry=0xbbbc90, pQuery=..., autodestroy=autodestroy@entry=false)
    at Database.cpp:341
#18 0x00000000004c6d78 in CDatabase::Write (this=0xbbbc90, pQuery=...) at Database.cpp:232
#19 0x00000000004ca84b in CDatabase::Open (this=this@entry=0xbbbc90, pFile=..., attach=...,
    allocation_chunk_size=allocation_chunk_size@entry=52428800, p_single_user_mutex=p_single_user_mutex@entry=0xbbb990,
    p_lock_mutex=p_lock_mutex@entry=0xbb7da0, p_lock_count=0xbbb860, p_unlock_cond=0xbb8e70, p_params=...)
    at /root/Entware/staging_dir/toolchain-x86_64_gcc-8.4.0_glibc-2.27/x86_64-openwrt-linux-gnu/include/c++/8.4.0/ext/new_allocator.h:79
#20 0x00000000004d715f in CServer::getDatabase (this=this@entry=0xb76dc0, tid=<optimized out>,
    pIdentifier=pIdentifier@entry=23) at Server.cpp:917
[... cut ...]
#24 0x00000000004f0748 in main_fkt (argc=<optimized out>, argv=<optimized out>) at main.cpp:554
#25 0x00000000004f2d7d in main_fkt_catch (argc=<optimized out>, argv=<optimized out>) at main.cpp:130
#26 0x000000000082db01 in run_real_main (args=...) at urbackupserver/cmdline_preprocessor.cpp:98
#27 0x00000000008396e7 in action_run (args=...) at urbackupserver/cmdline_preprocessor.cpp:531
#28 0x00000000004bfd3d in main (argc=<optimized out>, argv=0x7fffffffe898) at urbackupserver/cmdline_preprocessor.cpp:1270
(gdb)
```

zyxmon, 2022-01-31:
I have successfully run urbackup server on my Qnap NAS. I guess your architecture is x64. So Iэму used qnap TS-253b with x64 cpu. Here is the log
```
root@QNAP253:/share/CACHEDEV1_DATA/.qpkg/Entware/var$ urbackupsrv run -u root
2022-01-31 13:40:44: Starting HTTP-Server on port 55414
2022-01-31 13:40:44: HTTP: Server started up successfully!
2022-01-31 13:40:44: Generating Server identity...
2022-01-31 13:40:44: Generating Server private/public ECDSA key...
2022-01-31 13:40:44: Calculating public key...
2022-01-31 13:40:44: Generating Server private/public DSA key...
2022-01-31 13:40:44: Calculating public key...
2022-01-31 13:40:44: Generating Server token...
2022-01-31 13:40:44: WARNING: Upgrading...
2022-01-31 13:40:44: WARNING: Upgrading database to version 8
2022-01-31 13:40:45: WARNING: Upgrading database to version 9
2022-01-31 13:40:46: WARNING: Upgrading database to version 10
2022-01-31 13:40:46: WARNING: Upgrading database to version 11
2022-01-31 13:40:46: WARNING: Upgrading database to version 12
2022-01-31 13:40:46: WARNING: Upgrading database to version 13
2022-01-31 13:40:46: WARNING: Upgrading database to version 14
2022-01-31 13:40:46: WARNING: Upgrading database to version 15
2022-01-31 13:40:47: WARNING: Upgrading database to version 16
2022-01-31 13:40:47: WARNING: Upgrading database to version 17
2022-01-31 13:40:47: WARNING: Upgrading database to version 18
2022-01-31 13:40:47: WARNING: Upgrading database to version 19
2022-01-31 13:40:47: WARNING: Upgrading database to version 20
2022-01-31 13:40:48: WARNING: Upgrading database to version 21
2022-01-31 13:40:48: WARNING: Upgrading database to version 22
2022-01-31 13:40:48: WARNING: Upgrading database to version 23
2022-01-31 13:40:48: WARNING: Upgrading database to version 24
2022-01-31 13:40:48: WARNING: Upgrading database to version 25
2022-01-31 13:40:48: WARNING: Upgrading database to version 26
2022-01-31 13:40:48: WARNING: Upgrading database to version 27
2022-01-31 13:40:49: WARNING: Upgrading database to version 28
2022-01-31 13:40:49: WARNING: Upgrading database to version 29
2022-01-31 13:40:49: WARNING: Upgrading database to version 30
2022-01-31 13:40:49: WARNING: Upgrading database to version 31
2022-01-31 13:40:49: WARNING: Upgrading database to version 32
2022-01-31 13:40:50: WARNING: Upgrading database to version 33
2022-01-31 13:40:50: WARNING: Upgrading database to version 34
2022-01-31 13:40:50: WARNING: Upgrading database to version 35
2022-01-31 13:40:50: WARNING: Upgrading database to version 36
2022-01-31 13:40:50: WARNING: Upgrading database to version 37
2022-01-31 13:40:50: WARNING: Upgrading database to version 38
2022-01-31 13:40:51: WARNING: Upgrading database to version 39
2022-01-31 13:40:51: WARNING: Upgrading database to version 40
2022-01-31 13:40:51: WARNING: Upgrading database to version 41
2022-01-31 13:40:51: WARNING: Upgrading database to version 42
2022-01-31 13:40:51: WARNING: Upgrading database to version 43
2022-01-31 13:40:51: WARNING: Upgrading database to version 44
2022-01-31 13:40:52: WARNING: Upgrading database to version 45
2022-01-31 13:40:52: WARNING: Upgrading database to version 46
2022-01-31 13:40:52: WARNING: Upgrading database to version 48
2022-01-31 13:40:53: WARNING: Upgrading database to version 49
2022-01-31 13:40:53: WARNING: Upgrading database to version 50
2022-01-31 13:40:53: WARNING: Upgrading database to version 51
2022-01-31 13:40:53: WARNING: Upgrading database to version 52
2022-01-31 13:40:53: WARNING: Upgrading database to version 53
2022-01-31 13:40:53: WARNING: Upgrading database to version 54
2022-01-31 13:40:53: WARNING: Upgrading database to version 55
2022-01-31 13:40:54: WARNING: Upgrading database to version 56
2022-01-31 13:40:54: WARNING: Upgrading database to version 57
2022-01-31 13:40:54: WARNING: Upgrading database to version 58
2022-01-31 13:40:54: WARNING: Upgrading database to version 59
2022-01-31 13:40:54: WARNING: Done.
2022-01-31 13:40:55: Deleting database journal...
2022-01-31 13:40:55: Copying/reflinking database...
2022-01-31 13:40:55: Reflink ioctl failed. errno=95
2022-01-31 13:40:55: WARNING: Creating file entry index. This might take a while...
2022-01-31 13:40:55: Getting number of files...
2022-01-31 13:40:55: Dropping index...
[... 21 lines cut ...]
2022-01-31 13:40:55: Broadcasting on ipv4 interface vethgw01 addr 253.253.253.253
2022-01-31 13:40:55: Broadcasting on ipv6 interface br0 addr 2a02:2168:86bc:6b00:265e:beff:fe0e:d7c0
2022-01-31 13:40:55: Broadcasting on ipv6 interface br0 addr fe80::265e:beff:fe0e:d7c0
2022-01-31 13:40:55: Broadcasting on ipv6 interface docker0 addr fe80::5c17:62ff:fe6f:f7e0
2022-01-31 13:40:55: Broadcasting on ipv6 interface lxcbr0 addr fe80::7058:6bff:fe82:f8ef
2022-01-31 13:40:55: UrBackup Server start up complete.
2022-01-31 13:40:55: Looking for old Sessions... 0 sessions
2022-01-31 13:40:55: Server started up successfully!
2022-01-31 13:40:56: Downloading version file...
2022-01-31 13:40:57: Downloading signature...
2022-01-31 13:40:57: Downloading old signature...
2022-01-31 13:40:57: Getting update file URL...
2022-01-31 13:40:57: Downloading update file...
2022-01-31 13:41:05: Successfully downloaded update file.
2022-01-31 13:41:05: Downloading version file...
2022-01-31 13:41:05: Downloading signature...
2022-01-31 13:41:06: Getting update file URL...
2022-01-31 13:41:06: Downloading update file...
2022-01-31 13:41:12: Successfully downloaded update file.
2022-01-31 13:41:12: Downloading server version info...
2022-01-31 13:41:12: Downloading dataplan database...

```
I can connect to my nas port 55414 and continue setup through http. The only difference with your setup is that I have used alternative installer and root user to run urbackupsrv. I have created 2 directories - /opt/var/urbackup and /share/CACHEDEV1_DATA/Public/urbackup for backups.
Try to create these directories and run server.

barnumbirr, 2022-01-31:
As described above, urBackup works just fine until the first client backup. I can connect to the web interface, tweak the settings, trigger manual backups. etc... Once the first client backup has happened (independent of file or image backup) has finished, urbackup segfaults.
The following directories were created during package installation and are not the culprit:

```
/share/ZFS530_DATA/.qpkg/Entware/var/urbackup
/share/ZFS530_DATA/.qpkg/Entware/share/urbackup
/share/ZFS19_DATA/data/backups/urbackup
```

zyxmon, 2022-01-31:
@barnumbirr - I'll try to backup something and test when I have time.
I have compiled a new version 2.4.15 for testing, may be it'll solve problems - https://bin.entware.net/x64-k3.2/test/urbackup-server_2.4.15-1_x64-3.2.ipk
Please try it.

barnumbirr, 2022-01-31:
@zyxmon Thanks, upgrading urBackup is one of the things I wanted to try, hence #761.
Unfortunately, this is an instance of `urbackupsrv` starting and going a bit further than usual before segfaulting again:

```bash
[/share/data] # urbackupsrv run --user admin
2022-01-31 15:20:38: Starting HTTP-Server on port 55414
2022-01-31 15:20:38: HTTP: Server started up successfully!
2022-01-31 15:20:39: SQLite: recovered 1443 frames from WAL file /share/ZFS530_DATA/.qpkg/Entware/var/urbackup/backup_server.db-wal code: 283
2022-01-31 15:20:41: SQLite: recovered 235605 frames from WAL file /share/ZFS530_DATA/.qpkg/Entware/var/urbackup/backup_server_files.db-wal code: 283
2022-01-31 15:20:41: SQLite: recovered 9 frames from WAL file /share/ZFS530_DATA/.qpkg/Entware/var/urbackup/backup_server_link_journal.db-wal code: 283
2022-01-31 15:20:41: SQLite: recovered 159 frames from WAL file /share/ZFS530_DATA/.qpkg/Entware/var/urbackup/backup_server_settings.db-wal code: 283
Segmentation fault
```
I can clearly see from the log and the gdb stracktrace that the problem seems to be `sqlite`  but that unfortuantely doesn't help me understand what goes wrong.

zyxmon, 2022-01-31:
Have you checked the integrity of sqlite database? It looks like it was somehow corrupted.

barnumbirr, 2022-01-31:
Fair question, my bad for not specifying this earlier. Both database related commands segfault as well:

```bash
[/share/data] # urbackupsrv -u admin repair-database
Segmentation fault
[/share/data] # urbackupsrv -u admin defrag-database
2022-01-31 20:35:18: Shutting down all database instances...
2022-01-31 20:35:18: Opening urbackup server database...
Segmentation fault
```

zyxmon, 2022-01-31:
Why not to check db with sqlite3 binary.

barnumbirr, 2022-01-31:
Here's the output when checking `backup_server.db` with 

```bash
[/share/ZFS530_DATA/.qpkg/Entware/var/urbackup] # sqlite3 backup_server.db
SQLite version 3.33.0 2020-08-14 13:23:32
Enter ".help" for usage hints.
sqlite> .tables
alert_script_params    clients_hist_id        moved_clients
alert_scripts          del_stats              orig_client_settings
assoc_images           log_data               restores
backup_images          logs                   tokens_on_client
backups                mail_queue             user_tokens
clients                misc                   users_on_client
clients_hist           mounted_backup_images
sqlite> PRAGMA integrity_check;
ok
sqlite> VACUUM;
sqlite> .exit
```

I also tried exporting/importing the DB but unfortunately, that doesn't seem to help anything:

```bash
[/share/ZFS530_DATA/.qpkg/Entware/var/urbackup] # urbackupsrv run --user admin
2022-01-31 21:13:09: Starting HTTP-Server on port 55414
2022-01-31 21:13:09: HTTP: Server started up successfully!
Segmentation fault
```

zyxmon, 2022-01-31:
Here is a new version of sqlite3 library - https://bin.entware.net/x64-k3.2/test/libsqlite3_3370000-1_x64-3.2.ipk
may be it will help.

barnumbirr, 2022-01-31:
Unfortunately not, even breaks opening `backup_server.db` database

```bash
[~] # opkg install ./libsqlite3_3370000-1_x64-3.2.ipk
Upgrading libsqlite3 on root from 3330000-2 to 3370000-1...
Configuring libsqlite3.
[~] # urbackupsrv run --user admin
2022-01-31 22:25:13: Starting HTTP-Server on port 55414
2022-01-31 22:25:13: HTTP: Server started up successfully!
Segmentation fault
[~] cd /share/ZFS530_DATA/.qpkg/Entware/var/urbackup
[/share/ZFS530_DATA/.qpkg/Entware/var/urbackup] # sqlite3 backup_server.db
SQLite header and source version mismatch
2021-11-27 14:13:22 bd41822c7424d393a30e92ff6cb254d25c26769889c1499a18a0b9339f5d6c8a
2020-08-14 13:23:32 fca8dc8b578f215a969cd899336378966156154710873e68b3d9ac5881b0ff3f
```

barnumbirr, 2022-02-01:
I've now managed to get `gdb` running locally on the NAS and the error message thrown before the backtrace is a bit more informative:

```bash
Thread 1 "urbackupsrv" received signal SIGSEGV, Segmentation fault.
0x000000000055e04e in sqlite3WalFindFrame (pWal=0xbdfe28, pgno=1, piRead=piRead@entry=0x7fffffffb474) at sqlite/sqlite3.c:60697
60697   sqlite/sqlite3.c: No such file or directory.
(gdb)
```

@zyxmon what would be the best way to reinstall urbackup-server, while preserving it's config and data, to make sure it's properly linked to all it's dependencies?

barnumbirr, 2022-07-25:
Issue still persists, a few of my questions are still waiting for answers.

ryzhovau, 2023-05-23:
Looks like some device specific issue as long as we can't reproduce it.

closed 2023-05-23 by ryzhovau