Entware/Entware #682: Veracrypt - an excellent approach to encrypt and decrypt files

jmarks-lab, 2021-09-13:
Veracrypt (https://www.veracrypt.fr/en/Home.html) is a well-known, free open-source disk and file encryption software. 

Veracrypt is generally recognized as the standard in file encryption.

It is available for a variety of Linux distros, in both a console and GUI format. **This request is for an Entware package for the console version.**

It remains in active development - the latest version is August 7 2020, and there is an active discussion group attended by the creator Mounir Idrassi (https://sourceforge.net/p/veracrypt/discussion/)

Currently, backing up veractypt containers from a NAS is extremely time consuming, as the entire container (full of apparently random data) must be backed up, rather than the individual contents, which could constitute only 1 or 2 small files in a 1-2 GByte container.

Adding this package would allow veracrypt folders to be mounted and decrypted natively on the NAS, permitting file-by-file backups to other sites.

There been attempts at compiling a package for QNAP (https://www.synoforum.com/threads/compiling-veracrypt-on-your-nas.4707/).

I am prepared to test an Entware veracrypt package on both current and legacy QNAP NAs boxes.

Thank you!

zyxmon, 2021-09-13:
veracrypt needs wxwidgets 3.0 or higher. We have only 2.8 version.
Probably we will try to add veracrypt later but not this time.

closed 2021-09-13 by zyxmon

jmarks-lab, 2021-09-13:
Thank you for getting back so quickly.

Is the wxwidgets upgrade planned for a particular time?

Thank you.

jmarks-lab, 2021-09-13:
Andrei,
Is it possible to make an Entware package of the terminal version of Veracrypt, which would not require any GUI components?
Many thanks for your thoughts.
jm

zyxmon, 2021-09-14:
wxwigets may be used as a general purpose library. It should be investigated.
Please note we have encfs:
```
# opkg list encfs
encfs - 1.9.5-1 - EncFS provides an encrypted filesystem in user-space. It runs without any
 special permissions and uses the FUSE library module to provide the filesystem
 interface.

```
It may be similar to veracrypt.