Entware/Entware #686: Please add Perl DBD::PG Package

Digioso, 2021-09-17:
Hi,

I'm using Entware on my Synology DS920+ (x86_64 architecture).
root@diginas:~# uname -a
Linux diginas 4.4.180+ #41890 SMP Thu Jul 15 03:37:40 CST 2021 x86_64 GNU/Linux synology_geminilake_920+
root@diginas:~#

PostgreSQL is already installed on the box. Since I'm a Perl programmer I'd like to use it to connect to the PostgreSQL databases.
I'm not  really sure what exactly you need to add the package.

https://archlinux.org/packages/community/x86_64/perl-dbd-pg/

On Ubuntu the package is called libdbd-pg-perl and can be installed with apt-get.
On Red Hat / CentOS the package is called perl-DBD-Pg and can be installed via yum.

Unfortunately there is no real alternative. A pure Perl Perl Postgre Package exists, but from what I read that doesn't support Unicode. So not really an alternative when you're working with Unicode data... I tried i out nevertheless and it didn't work for me.

zyxmon, 2021-09-18:
Have you tryed this - https://github.com/Entware/Entware/wiki/Self-installation-of-perl-modules

Digioso, 2021-09-18:
> 
> 
> Have you tryed this - https://github.com/Entware/Entware/wiki/Self-installation-of-perl-modules

Thanks a lot! I tried just it, but unfortunately it doesn't work.
Basically CPAN is doing the same.

I followed the prerequisites and tried the manual installation step for DBD::Pg

```
root@diginas:/volume1/andi/DBD-Pg-3.15.0# perl Makefile.PL
Configuring DBD::Pg 3.15.0
Path to pg_config?
No POSTGRES_HOME defined, cannot find automatically
root@diginas:/volume1/andi/DBD-Pg-3.15.0#
```

As stated above there is no pg_config file on my system. I tried it with defining POSTGRES_HOME manually to what I think may be the correct directory (/var/services/pgsql).
But the next variable it looks for is POSTGRES_INCLUDE. This points to a include directory below POSTGRES_HOME. However in my case there is no include directory there. I searched the whole box (find / -name include) and there is no include directory that has anything to do with Postgres.

Digioso, 2021-09-18:
Looking further it seems that pg_config may have to be installed manually. According to stackoverflow and other sites this is in the libpq-dev or postgresql-devel packages (depending on the distro). Both are not available with Entware unfortunately. :(

zyxmon, 2021-09-18:
There are no dev packages in Entware. Please read the wiki. Sometimes patches are needed for compilation because of nonstandard (/opt) files locations.

Digioso, 2021-09-18:
That's why I am asking to add a DBD::pg package directly. Then no dev packages are required.

closed 2021-09-28 by The-BB