Entware/Entware #688: Segmentation fault - pixelserv-tls

wellloaded, 2021-09-18:
**Describe the bug**
```
opkg install pixelserv-tls

/opt/bin/pixelserv-tls 192.168.0.200 -f
Segmentation fault

```
**Environment (please complete the following information):**
 - Entware feed: 
```
root@router:/tmp/home/root# opkg print-architecture
arch all 100
arch armv7soft 150
arch armv7-2.6 160
```

 - Device Asus RT-AC66U_B1
 - Firmware version FreshTomato 2021.5

zyxmon, 2021-09-19:
Confirmed - package crashes also on mipsel and armv7-3.2.

zyxmon, 2021-09-19:
Have you generated certificates in /opt/var/cache/pixelserv ? No segfault with certificates in the correct dir.

wellloaded, 2021-09-19:
Spot on! I thought by default it would open port 80 but it actually try to open 443 and expects certificates (by default).

Perhaps this can be better documented on the developer side but yes false positive for the entware issue tracker, thanks for helping!

closed 2021-09-19 by zyxmon