Entware/Entware #690: Please whenever is possible update Tailsale to version -  tailscale_1.14.3

Alozaros, 2021-09-22:
A new tailscale update is available, could you update it, if possible ?
I'm trying to make a use of it, as a service and create a guide on DDWRT platform .
Thanks in advance!
https://pkgs.tailscale.com/stable/

- [ ] No similar package already added and/or this one got some new killer feature(s) - yes it contains some fixes and updates
- [ ] Package is in active development - yes
- [ ] I'm ready to test added package on my platform [armv7sf-k3.2] and get a feedback about it.

zyxmon, 2021-09-23:
Here are test packages for your architecture. Please give feedback.
A guide on DDWRТ platform would be nice. Please post URL here when it is ready.

Alozaros, 2021-09-25:
https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=330217
still WIP as im testing/fiddling it, but its fully working in this stage...
By any chance could i have tailscale_1.14.3 for mipssf-k3.4 too
Thanks in advance!

closed 2021-09-25 by ryzhovau

ryzhovau, 2021-09-25:
mipsel binary updated, the others will arrive late.