Entware/Entware #694: mkvtoolnix errors

tleepa, 2021-09-30:
**Description**
[mkvtoolnix 58.0.0-1]: Running any binary from this package (mkvinfo, mkvextract, mkvmerge) errors with message:

```shell
mkvbinary: symbol lookup error: mkvbinary: undefined symbol: _ZTIN3fmt2v712format_errorE
```

It may be an issue with fmt version. Current version of fmt on my NAS is `fmt (GNU coreutils) 8.24`.

MKVToolnix's [changelog](https://www.videohelp.com/software/MKVToolNix/version-history) mentions fixing fmt-related ["Build against fmt 8.0.0 fails" issue](https://gitlab.com/mbunkus/mkvtoolnix/-/issues/3151) in version 59.0.0.

So, maybe updating mkvtoolnix package to the latest version will solve the problem.

**Environment**
 - Entware feed: armv7sf-k2.6
 - Device: Synology DS214 NAS
 - Firmware version: DSM 6.2.4-25556 Update 2

zyxmon, 2021-09-30:
It looks like some of the libraries was not updated. We have a hint in the wiki regarding this problem.
mrvtoolnix is running fine on ma synology NAS. Probably this will help

$ mkvinfo -V
mkvinfo v58.0.0 ('Supper's Ready') 32-bit
root@QNAPD1:/share/CACHEDEhis will help

```
$ mkvinfo -V
mkvinfo v58.0.0 ('Supper's Ready') 32-bit
$ ldd /opt/bin/mkvinfo
        libstdc++.so.6 => /opt/lib/libstdc++.so.6 (0xb6e67000)
        libintl.so.8 => /opt/lib/libintl.so.8 (0xb6e4e000)
        libmagic.so.1 => /opt/lib/libmagic.so.1 (0xb6e1c000)
        libz.so.1 => /opt/lib/libz.so.1 (0xb6df8000)
        libFLAC.so.8 => /opt/lib/libFLAC.so.8 (0xb6dbc000)
        libogg.so.0 => /opt/lib/libogg.so.0 (0xb6da5000)
        libiconv.so.2 => /opt/lib/libiconv.so.2 (0xb6cb6000)
        libfmt.so.8 => /opt/lib/libfmt.so.8 (0xb6c8b000)
        libpcre2-8.so.0 => /opt/lib/libpcre2-8.so.0 (0xb6bf7000)
        libm.so.6 => /opt/lib/libm.so.6 (0xb6b32000)
        libssp.so.0 => /opt/lib/libssp.so.0 (0xb6b1f000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0xb6b02000)
        libc.so.6 => /opt/lib/libc.so.6 (0xb69c0000)
        /opt/lib/ld-linux.so.3 (0xb6fb5000)
        liblzma.so.5 => /opt/lib/liblzma.so.5 (0xb6998000)
        libbz2.so.1.0 => /opt/lib/libbz2.so.1.0 (0xb6979000)
        libpthread.so.0 => /opt/lib/libpthread.so.0 (0xb6950000)
```

tleepa, 2021-09-30:
Well, it is looking a bit different here:
```
❯ ldd /opt/bin/mkvinfo
        not a dynamic executable
```

But! I did something I rarely do on Linux systems - reinstalled mkvtoolnix package and it works!

Case closed :)

closed 2021-09-30 by ryzhovau