Entware/Entware #696: Tailscale: tailscale and tailscaled both exhibit 'illegal instruction' when executed on armv7sf-k2.6

davygravy, 2021-10-03:
**Description:**
tailscale: tailscale and tailscaled both exhibit 'illegal instruction' when executed armv7sf-k2.6. 

**Environment:**
 - Entware feed: [armv7sf-k2.6]
 - Device [Linksys EA6900 and Netgear R6250 Routers]
 - Firmware version [Fresh Tomato ARM v2021.5]

**Comments**
 - perhaps compiler flag(s) not set correctly for this package
 - N.B. wireguard (`wg`) does seem to execute correctly - though I haven't tested its operation.  I mention this because `tailscale` is essentially a zeroconfig tool for wireguard.
 - I am **certain** that tailscale works on some other ARM devices (e.g. Kirkwood boxes) because I've tested it and I use it.

**Behavior**
Simple execution:
```
root@AirGate:/usr/sbin# tailscale
Illegal instruction
root@AirGate:/usr/sbin# tailscaled
Illegal instruction
```

Using `strace`:
```
root@AirGate:/usr/sbin# strace tailscale
execve("/opt/bin/tailscale", ["tailscale"], 0xbed21d10 /* 18 vars */) = 0
getpid()                                = 31244
--- SIGILL {si_signo=SIGILL, si_code=ILL_ILLOPC, si_addr=0x5a6e8} ---
+++ killed by SIGILL +++
Illegal instruction
root@AirGate:/usr/sbin# strace tailscaled
execve("/opt/bin/tailscaled", ["tailscaled"], 0xbea64d10 /* 18 vars */) = 0
getpid()                                = 31533
--- SIGILL {si_signo=SIGILL, si_code=ILL_ILLOPC, si_addr=0x5d174} ---
+++ killed by SIGILL +++
Illegal instruction
```

zyxmon, 2021-10-03:
Probably you need to install nohf versions of go packages.

davygravy, 2021-10-03:
You are correct.  Thank you and please close this issue.  If I see any further problem, I'll either reopen this or start a new one, as appropriate.

Thank you!

closed 2021-10-03 by davygravy

davygravy, 2023-01-30:
> @davygravy I encountered the same issue. Could please elaborate on how you resolved it? I am actually confused by what @zyxmon said install nohf versions of go packages. Was it a particular tailscale package I need to install with? Thanks in advance!

IIRC, I just had to install the version is the armv7 k26 Tailscale package that had “nohf” in its name.

davygravy, 2023-01-30:
Package name is: tailscale_nohf