Entware/Entware #697: Telegraf: Illegal instruction

satcom886, 2021-10-05:
**Bug description**
Install the `mipssf-k3.4` build of `telegraf` and try to run it. The result is the program immediately quitting and outputting:
```
Illegal instruction
```

**Environment**
 - Entware feed: mipssf-k3.4
 - Device: TP-Link WR1043ND
 - Firmware version: DD-WRT

satcom886, 2021-10-05:
Output of `cat /proc/cpuinfo`:

```
system type             : Qualcomm Atheros QCA9558 ver 1 rev 1.0 (0x1130)
processor               : 0
cpu model               : MIPS 74Kc V5.0
BogoMIPS                : 358.80
CPUClock                : 720
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16
Options implemented     : tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available
```

satcom886, 2021-10-05:
Ok, I also tested on an ARM router and it's the same thing.

```
model name      : ARMv7 Processor rev 0 (v7l)
BogoMIPS        : 1594.16
Features        : half fastmult edsp tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc09
CPU revision    : 0

Hardware        : Northstar Prototype
Revision        : 0000
Serial          : 0000000000000000
```

satcom886, 2021-10-05:
I just noticed there's a `nohf` build of `telegraf` for ARM. I installed that and it works now.

There isn't one for MIPS however. Makes sense since the `sf` in `mipssf` stands for soft-float (at least I think so 🙂). I suspect there's a wrong compiler flag somewhere. So that problem still remains.

ryzhovau, 2021-12-09:
> There isn't one for MIPS however. Makes sense since the sf in mipssf stands for soft-float (at least I think so 🙂). I suspect there's a wrong compiler flag somewhere. So that problem still remains.

DD-WRT users faced similar issues with other Entware packages. Can't help there.

closed 2021-12-09 by ryzhovau