Entware/Entware #707: Package request: ripgrep

Lightdisappear, 2021-10-21:
**Describe the solution you'd like**
> ripgrep is a line-oriented search tool that recursively searches the current directory for a regex pattern. By default, ripgrep will respect gitignore rules and automatically skip hidden files/directories and binary files.

[https://github.com/BurntSushi/ripgrep](https://github.com/BurntSushi/ripgrep)

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. x86-64] and get a feedback about it.

The-BB, 2021-11-05:
Please test it http://bin.entware.net/x64-k3.2/test/ripgrep_13.0.0-3_x64-3.2.ipk

closed 2021-11-19 by no-response[bot]

goetzc, 2022-09-03:
@ryzhovau, out of curiosity, why is this not possible? 

There are many great Rust applications out there, which would be very handy to have available as Entware packages.

ryzhovau, 2022-09-04:
A year ago it was harder to cross-compile. Now @The-BB is testing some Rust apps, so any feedback would be appreciated.

goetzc, 2022-09-05:
Oh so it was because of the cross-compilation, got it.

Are there any ARMv7sf-k2.6 packages to help test any of these out?