Entware/Entware #715: Please add PHP sodium package

nearPath, 2021-11-05:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

- [X] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

kimdoyeonsa, 2021-11-13:
php8 module sodium package
http://kimdoyeonsa.ipdisk.co.kr:9100/list/HDD1/packages/aarch64-3.10/
http://kimdoyeonsa.ipdisk.co.kr:9100/list/HDD1/packages/armv5-3.2/
http://kimdoyeonsa.ipdisk.co.kr:9100/list/HDD1/packages/armv7-2.6/
http://kimdoyeonsa.ipdisk.co.kr:9100/list/HDD1/packages/armv7-3.2/
http://kimdoyeonsa.ipdisk.co.kr:9100/list/HDD1/packages/mips-3.4/
http://kimdoyeonsa.ipdisk.co.kr:9100/list/HDD1/packages/mipsel-3.4/
http://kimdoyeonsa.ipdisk.co.kr:9100/list/HDD1/packages/x64-3.2/
http://kimdoyeonsa.ipdisk.co.kr:9100/list/HDD1/packages/x86-2.6/

ryzhovau, 2021-12-10:
Available only for PHP7 at this time.

closed 2021-12-10 by ryzhovau