Entware/Entware #716: opkg program does not allow changing lockfile location; cannot run if /opt is readonly

jtmoon79, 2021-11-07:
## Problem

An embedded Linux device with a readonly mount of path `/opt` means `opkg` fails `Could not create lock file /opt/tmp/opkg.lock`. The `opkg` installer should allow a different lockfile location, .e.g. `--lock path/opkg.lock`

### Reproduction

Given any system with readonly `/opt` path.
1. `cd /tmp`
2. Download [the `opkg` file](https://bin.entware.net/armv7sf-k3.2/installer/opkg).
3. Download [the `opkg.conf` file](https://bin.entware.net/armv7sf-k3.2/installer/opkg.conf).
4. Change `opkg.conf` file from
```
src/gz entware http://bin.entware.net/armv7sf-k3.2
dest root /
dest ram /opt/tmp
lists_dir ext /opt/var/opkg-lists
option tmp_dir /opt/tmp
arch all 100
arch armv7-3.2 160
```
to a writeable location (any writable path will do; just prepend `/tmp`)
```
src/gz entware http://bin.entware.net/armv7sf-k3.2
dest root /
dest ram /tmp/opt/tmp
lists_dir ext /tmp/opt/var/opkg-lists
option tmp_dir /tmp/opt/tmp
arch all 100
arch armv7-3.2 160
```
5. run `opkg`
```
$ /tmp/opkg --conf /tmp/opkg.conf -V3 -d /tmp/ -t /tmp/ update
opkg_conf_parse_file: Loading conf file /tmp/opkg.conf.
opkg_conf_parse_file: Supported arch all priority (100)
opkg_conf_parse_file: Supported arch armv7-3.2 priority (160)
Collected errors:
 * opkg_conf_set_option: Duplicate option tmp_dir, using first seen value "/tmp/".
 * opkg_conf_load: Could not create lock file /opt/tmp/opkg.lock: No such file or directory.
```
The culprit message is `Could not create lock file /opt/tmp/opkg.lock: No such file or directory`.
On my system, the path `/opt` is readonly (because the mount root `/` is readonly).

## Solution

The `opkg` should have an option to specify the lockfile path.
```
$ /tmp/opkg --lock /tmp/opkg.lock
```
The `opkg.conf` should also allow specifying the lockfile path.

In my case, I could force the root path `/` to be read-write. However, I'd prefer `opkg` offer lock file flexibility.

### Environment
 - Entware feed: `armv7sf-k3.2`
 - Firmware version DDWRT latest

zyxmon, 2021-11-07:
Entware will not work if installed (moved) to a folder different from `/opt`.
Lockfile location is defined at build time - https://github.com/Entware/Entware/blob/master/package/system/opkg/Makefile#L71

You can try `offline-root`  option with opkg. But I do not think it will help you.

zyxmon, 2021-11-07:
May be this will help - https://wiki.dd-wrt.com/wiki/index.php/Installing_Entware

closed 2021-11-07 by zyxmon

jtmoon79, 2021-11-07:
> Lockfile location is defined at build time

@zyxmon is it possible to add a command-line option to set that path value?

In my case, the `/opt` path has files that were unsquashed at boot time.

thinklover, 2022-04-30:
I also have the same problem.
I hope that we can chang lockfile path.

thinklover, 2022-04-30:
> May be this will help - https://wiki.dd-wrt.com/wiki/index.php/Installing_Entware

I also have the same problem.
I hope that we can chang lockfile path.

patrickelectric, 2022-05-24:
The system that I'm working on has / as a read-only and it's impossible to create the /opt folder. I was able to have opkg but having problems with the lock file!

zyxmon, 2022-05-25:
Entware will work **only** from `/opt` folder.  In case you cannot create /opt - you can run entware in chroot environment
https://github.com/Entware/Entware/issues/775