Entware/Entware #719: Poor openconnect performance

TTimo, 2021-11-11:
I am using openconnect from the entware repository on a RT-AX88U router running asuswrt-merlin:

```
OpenConnect version v8.10
Using GnuTLS 3.7.1. Features present: HOTP software token, TOTP software token, System keys, DTLS, ESP
Supported protocols: anyconnect (default), nc, gp, pulse
```

However, the performance is terrible (10Mbps download speeds when I should be getting 80 or more).

I talked with the auswrt a long time ago about this, and we suspect this is because gnutls does not take advantage of hardware acceleration.

A build of openconnect that uses openssl instead may have dramatically better performance (assuming we can make it use the OS image's `/usr/lib/libssl.so.1.1`)

Could we get an alternate package that uses openssl?

zyxmon, 2021-11-12:
We do not build Entware with hardware acceleration support. The is an cortex-15 version of Entware tested on netgear qualcom routers here - http://cortex-a15.zyxmon.org/binaries/cortex-a15-3x/ - it has hardware acceleration support.

closed 2021-11-12 by zyxmon

TTimo, 2021-11-14:
Hello @zyxmon - is there documentation showing how the cortex-a15 packages may be installed?

I tried to add the architecture to my entware configuration, but I'm getting nowhere:

```
src/gz entware http://bin.entware.net/aarch64-k3.10
src/gz zyxmon http://cortex-a15.zyxmon.org/binaries/cortex-a15-3x
dest root /
dest ram /opt/tmp
lists_dir ext /opt/var/opkg-lists
option tmp_dir /opt/tmp
arch all 100
arch aarch64-3.10 160
arch cortex-a15-3x 200
```

Packages that I replace with their `cortex-a15-3x` version simply don't work. The error is strange to me, but I suspect it means the binary format is not supported:

```
admin@RT-AX88U-9B60:/tmp/mnt/router/entware.cortex-a15-3x.1/sbin# ls -1al openconnect
-rwxr-xr-x    1    285668 Feb 28  2021 openconnect
admin@RT-AX88U-9B60:/tmp/mnt/router/entware.cortex-a15-3x.1/sbin# ./openconnect
-sh: ./openconnect: not found
```

zyxmon, 2021-11-14:
Just reinstall Entware using new repository URL.
Please check if your CPU is cortex-a15 compatible before reinstallation.

TTimo, 2021-11-15:
This is a BCM4908 - I'm not very familiar with embedded systems, but I think this is not compatible with the cortex-a15 repository.
e.g. BCM4908 is a cortex-b53, aka aarch64 (64 bit armv8), and cortex-a15 is armv7 32 bit.

So I'm back with my problem - I need an aarch64 openconnect binary that is ABI compatible with asuswrt-merlin's libssl library, which I believe has all the hardware acceleration I'm going to need to get good performance.

I would like to have a github task open so I can have a bounty on it and hopefully find someone interested in helping me. If you don't want to leave tasks open against the entware repository, maybe you can suggest somewhere else I can find someone to help?

zyxmon, 2021-11-15:
I still suggest you to try cortex-15 version of Entware. It may will run on aarch64 hardware - this depends on the firmware kernel.
Voxel - https://www.voxel-firmware.com/Downloads/Voxel/html/index.html is the author of hardware accelerate Entware. You can find him on snmforums - he can tell what modifications he uses to build Entware.