Entware/Entware #720: Please add "nfs-ganesha" package

tuhajbej6, 2021-11-17:
files:
 ganesha-nfsd executable and dependable libraries

This might be evolutionary for aarch64 devices

ryzhovau, 2021-12-09:
We've got another userspace NFS server https://github.com/unfs3/unfs3
What's the killer feature of `ganesha-nfsd`?

tuhajbej6, 2021-12-09:
Hi, the killer features I guess are following:
- character translation of filenames (NFS version 4 provides filename
character translation over version 3 which doesn't)
- possibly ReadWrite possible on Windows (ReadOnly is maximum what I get
whatever the export & mount combination is - although deletion is also
possible but not recursive )
- still it's next generation 4 vs 3

Cheers.
T.

ryzhovau, 2021-12-10:
Can't get it work on MIPS target with Linus kernel 4.9. Export list is always empty:
```
nfs_Init :NFS STARTUP :INFO :NFSv4 ACL cache successfully initialized
init_export_root :EXPORT :DEBUG :About to lookup_path for ExportId=12345 Path=/opt
vfs_lookup_path :FSAL :DEBUG :filesystem /tmp/mnt/f6efbdd3-1f47-d601-b02f-b8d31f47d601 for path /opt
vfs_map_name_to_handle_at :FSAL :DEBUG :Error Function not implemented (89) bytes = 48
vfs_lookup_path :FSAL :CRIT :Could not get handle for path /opt, error Function not implemented
posix2fsal_error :FSAL :CRIT :Default case mapping Function not implemented (89) to ERR_FSAL_SERVERFAULT
mdcache_lookup_path :INODE :DEBUG :lookup_path /opt failed with Undefined server error
init_export_root :EXPORT :CRIT :Lookup failed on path, ExportId=12345 Path=/opt FSAL_ERROR=(Undefined server error,89)
init_export_root :EXPORT :DEBUG :About to lookup_path for ExportId=0 Path=/
```
Looks like `nfs-ganesha` is relying on some syscalls, which is missing on my system.

closed 2021-12-10 by ryzhovau