Entware/Entware #723: no /dev/btrfs-control

armandckj1, 2021-11-24:
**Describe the bug**
[package name]: A clear and concise description of what the bug is and how to reproduce it. 
Btrfs-progs

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4]
 - Device [e.g. Asus RT-AX88U router]
 - Firmware version [e.g. asuswrt-merlin]
Netgear R8000-FT 
Freshtomato 2021.7 (error first noticed on 2021.5)
Entware Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz

I use a usb disk on the netgear for backup purposes with rsync.

Installed btrfs-progs
When trying 'btrfs dev scan'  I get WARNING: failed to open /dev/btrfs-control, skipping device registration: No such device

https://btrfs.wiki.kernel.org/index.php/Problem_FAQ#I_get_the_message_.22failed_to_open_.2Fdev.2Fbtrfs-control_skipping_device_registration.22_from_.22btrfs_dev_scan.22
gives a solution
mknod /dev/btrfs-control c 10 234
and You might also want to report to your distribution that their configuration without udev is missing this device.

Unfortunately creating 
crw-r--r--    1 root     root       10, 234 Nov 24 06:31 btrfs-control
does not solve the problem.
also chmod 666 does not help.

New options?

zyxmon, 2021-11-24:
This is kernel issue. You can contact firmware manufacturer on.
This cannot be solved in Entware. Sorry.

closed 2021-11-24 by zyxmon

armandckj1, 2021-11-24:
oke.
Thanks!