Entware/Entware #725: pkg socksify: `wrong ELF class: ELFCLASS64`

HUMORCE, 2021-11-30:
**Describe the bug**
[socksify]: does not work properly.

reproduction:

```shell
> opkg install socksify
> vi /opt/etc/socks.conf
...
> sockify curl example.org
ERROR: ld.so: object 'libdsocks.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
```

**Environment:**
 - Entware feed: aarch64-k3.10
 - Device: Asus RT-AX88U / RT-AC86U
 - Firmware version: asuswrt-merlin

zyxmon, 2021-12-01:
https://github.com/Entware/Entware/issues/330
We will remove this package from repo.
I am keeping this issue opened and close it after package removal (with next sync).

HUMORCE, 2021-12-01:
what alternative pkgs are available?

ryzhovau, 2021-12-01:
LD_PRELOAD is a quite dirty hack, so any other socksifier will work as good (or as bad) as this one.

closed 2021-12-05 by HUMORCE