Entware/Entware #730: Bad interpreter "/opt/bin/sh"

rustrict, 2021-12-11:
Привет! При одновременном обновлении busybox'a и terminfo у последней на этапе "Configuring" появляется `Bad interpreter "/opt/bin/sh"`.

Не уверен, что правильно понимаю работу `opkg upgrade`, поэтому не знаю, поможет ли [здесь](https://github.com/Entware/Entware/blob/master/package/libs/ncurses/Makefile#L169) замена shebang на `#!/opt/bin/busybox sh`.

Entware feed: mipselsf-k3.4
Device: Keenetic Giga (KN-1010)

zyxmon, 2021-12-12:
shebang в указанном месте заменять не надо. Считается, что Entware устанавливается на устройства, где есть shell /bin/sh. Есть два полу-исключения из этого правила - роутеры кинетик и устройства на android.
Приведите лог `opkg upgrade` для изучения ошибки.

rustrict, 2021-12-12:
К сожалению, со своего роутера не сохранил. Это лог с чужого Keenetic Ultra (KN-1810):
```shell
~ # opkg upgrade
Upgrading terminfo on root from 6.2-2a to 6.2-3...
Downloading http://bin.entware.net/mipselsf-k3.4/terminfo_6.2-3_mipsel-3.4.ipk
Upgrading zoneinfo-europe on root from 2021a-1 to 2021e-1...
Downloading http://bin.entware.net/mipselsf-k3.4/zoneinfo-europe_2021e-1_mipsel-3.4.ipk
Upgrading busybox on root from 1.33.1-1 to 1.34.1-1...
Downloading http://bin.entware.net/mipselsf-k3.4/busybox_1.34.1-1_mipsel-3.4.ipk
Upgrading glib2 on root from 2.68.1-3 to 2.70.0-1...
Downloading http://bin.entware.net/mipselsf-k3.4/glib2_2.70.0-1_mipsel-3.4.ipk
Removing obsolete file /opt/lib/libgthread-2.0.so.0.6800.1.
Removing obsolete file /opt/lib/libgobject-2.0.so.0.6800.1.
Removing obsolete file /opt/lib/libgio-2.0.so.0.6800.1.
Removing obsolete file /opt/lib/libglib-2.0.so.0.6800.1.
Removing obsolete file /opt/lib/libgmodule-2.0.so.0.6800.1.
Upgrading libpcre on root from 8.45-1 to 8.45-2...
Downloading http://bin.entware.net/mipselsf-k3.4/libpcre_8.45-2_mipsel-3.4.ipk
Upgrading grep on root from 3.6-1a to 3.7-2...
Downloading http://bin.entware.net/mipselsf-k3.4/grep_3.7-2_mipsel-3.4.ipk
Upgrading zoneinfo-asia on root from 2021a-1 to 2021e-1...
Downloading http://bin.entware.net/mipselsf-k3.4/zoneinfo-asia_2021e-1_mipsel-3.4.ipk
Upgrading libffi on root from 3.4.2-1 to 3.4.2-2...
Downloading http://bin.entware.net/mipselsf-k3.4/libffi_3.4.2-2_mipsel-3.4.ipk
Upgrading libubox on root from 2021-05-16-b14c4688-1 to 2021-11-20-cce5e351-1...
Downloading http://bin.entware.net/mipselsf-k3.4/libubox_2021-11-20-cce5e351-1_mipsel-3.4.ipk
Upgrading opt-ndmsv2 on root from 1.0-15 to 1.0-16...
Downloading http://bin.entware.net/mipselsf-k3.4/keenetic/opt-ndmsv2_1.0-16_mipsel-3.4_kn.ipk
Upgrading poorbox on root from 1.33.1-1 to 1.34.1-1...
Downloading http://bin.entware.net/mipselsf-k3.4/keenetic/poorbox_1.34.1-1_mipsel-3.4_kn.ipk
Upgrading bind-libs on root from 9.17.13-1 to 9.17.20-1...
Downloading http://bin.entware.net/mipselsf-k3.4/bind-libs_9.17.20-1_mipsel-3.4.ipk
Removing obsolete file /opt/lib/libdns-9.17.13.so.
Removing obsolete file /opt/lib/libisccfg-9.17.13.so.
Removing obsolete file /opt/lib/libisc-9.17.13.so.
Removing obsolete file /opt/lib/libirs-9.17.13.so.
Removing obsolete file /opt/lib/libbind9-9.17.13.so.
Removing obsolete file /opt/lib/libns-9.17.13.so.
Removing obsolete file /opt/lib/libisccc-9.17.13.so.
Upgrading libncurses on root from 6.2-2a to 6.2-3...
Downloading http://bin.entware.net/mipselsf-k3.4/libncurses_6.2-3_mipsel-3.4.ipk
Upgrading libncursesw on root from 6.2-2a to 6.2-3...
Downloading http://bin.entware.net/mipselsf-k3.4/libncursesw_6.2-3_mipsel-3.4.ipk
Upgrading htop on root from 3.0.5-1a to 3.1.0-1...
Downloading http://bin.entware.net/mipselsf-k3.4/htop_3.1.0-1_mipsel-3.4.ipk
Upgrading ca-bundle on root from 20210119-1 to 20211016-1...
Downloading http://bin.entware.net/mipselsf-k3.4/ca-bundle_20211016-1_all.ipk
Upgrading bind-dig on root from 9.17.13-1 to 9.17.20-1...
Downloading http://bin.entware.net/mipselsf-k3.4/bind-dig_9.17.20-1_mipsel-3.4.ipk
Upgrading libopenssl on root from 1.1.1k-1 to 1.1.1l-1...
Downloading http://bin.entware.net/mipselsf-k3.4/libopenssl_1.1.1l-1_mipsel-3.4.ipk
Configuring terminfo.
Bad interpreter "/opt/bin/sh"
Configuring libpcre.
Configuring grep.
Configuring zoneinfo-europe.
Configuring libubox.
Configuring busybox.
Warning: '/opt/bin/grep' is already exists, skipping.
Warning: '/opt/bin/crontab' is already exists, skipping.
Warning: '/opt/bin/fgrep' is already exists, skipping.
Warning: '/opt/bin/find' is already exists, skipping.
Warning: '/opt/bin/egrep' is already exists, skipping.
Warning: '/opt/bin/run-parts' is already exists, skipping.
Warning: '/opt/bin/xargs' is already exists, skipping.
Configuring libffi.
Configuring glib2.
Configuring libopenssl.
Configuring ca-bundle.
Configuring zoneinfo-asia.
Configuring opt-ndmsv2.
Updating /opt/etc/ld.so.cache... done.
Configuring poorbox.
Configuring bind-libs.
Configuring libncursesw.
Configuring libncurses.
Configuring htop.
Configuring bind-dig.
Collected errors:
 * resolve_conffiles: Existing conffile /opt/etc/passwd is different from the conffile in the new package. The new conffile will be placed at /opt/etc/passwd-opkg.
```
Как я понимаю, проблема из-за того, что на этапе `Upgrading busybox on root from 1.33.1-1 to 1.34.1-1...` скрипт `/opt/lib/opkg/info/busybox.prerm` сносит симлинк `/opt/bin/sh`, а `terminfo.postinst` отрабатывает до восстановления симлинков `busybox.postinst`.

Если я прав, на кинетиках любой `.postinst` с кодом и shebang `#!/bin/sh`, который выполнится до `busybox.postinst`, выкинет `Bad interpreter`.

ryzhovau, 2021-12-12:
Для этих целей в кинетиках ставится `poorbox` — урезанный shell в `/opt/usb/bin`, необходимый исключительно на время обновления busybox. 

Коль обсуждаем firmware specific случай, прошу переместиться на форум кинетиков. Мы там тоже присутсвуем.

closed 2021-12-12 by ryzhovau