Entware/Entware #732: Request GNU gettext

VA1DER, 2021-12-12:
**Describe the solution you'd like**
Request GNU gettext and its accompanying tools (ie: msgfmt).  This utility is ubiquitous in virtually any environment where you want to compile almost anything.

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2021-12-13:
No plans to develop native compilation tools, sorry.

closed 2021-12-13 by ryzhovau

VA1DER, 2021-12-13:
Not asking for them to be developed.  Asking for the [GNU gettext](https://www.gnu.org/software/gettext/) tools to be compiled into an Entware package.  Not sure I understand the reasoning behind providing Entware packages for go and gcc (including a nice set of include files for gcc) if native compilation tools are out.