Entware/Entware #733: openssh connection error

kimdoyeonsa, 2021-12-13:
**Describe the bug**
[package name]: A clear and concise description of what the bug is and how to reproduce it. 
Remote side unexpectedly closed network connection 
[
[image]
](url)
 line 232: -> $(INSTALL_BIN) $(PKG_INSTALL_DIR)/opt/sbin/sshd $(1)/opt/sbin/ssh

**Environment (please complete the following information):**
 - Entware feed: armv7sf-k3.2
 - Device iptime nas1dual 
 - Firmware version [e.g. asuswrt-merlin]

zyxmon, 2021-12-13:
Please decribу your problem with more details. I have just tried openssh-server on Qnap Nas (armv7sf-k3.2) without problems.

zyxmon, 2021-12-13:
My guess - your are using some 3rd party scripts that rely on busybox from Entware is installed.

kimdoyeonsa, 2021-12-13:
Loss of connection over time.

zyxmon, 2021-12-13:
Probably it is a problem with your device - may be some sleep mode. I've waited several houres with active ssh connection with my NAS. Connection is OK.

kimdoyeonsa, 2021-12-14:
My guess was that my nas was blocking the /sbin/sshd service. Renaming it to /sbin/ssh solved the problem.

closed 2021-12-14 by zyxmon