Entware/Entware #735: openssh-client: the name and path of binaries changed

Alnoox, 2021-12-14:
**Describe the bug**
[openssh-client]: After the last upgrade to version 8.8p1-2 (Dec 10 2021) the path to binaries (`ssh` and `scp`) changed from `/opt/bin/[ssh]|[scp]` to `/opt/libexec/[ssh]|[scp]-openssh`.

**Environment:**
 - Entware feed: armv7-3.2
 - Device: Synology DS218j
 - Firmware version: DSM 6.2.1

**Additonal info:**

First of all, thank you for this project and I hope this is the right place for this issue.
This new version of `openssh-client` changed these paths and I could not find any explanation or any warning that this would happen. It's quite problematic for me since I had packages with `openssh-client` as a dependency so it could use an SSH client. Now it won't work because `/opt/libexec` is not a directory included in the `PATH` variable and the binary's name changed too.

Am I doing something wrong? Can you confirm if it will stay like this or if changes will be made (like a symlink) in a next release? It would be useful to know if I can wait a bit or if I need to fix it on my devices eventually.

Thank you in advance.

VA1DER, 2021-12-18:
This appears to be a wider problem with several Entware packages.  For example, scp, less, and ssh are all affected.  Generally binaries that are installed into libexec have soft-links made to bin, and these are not being made.

ryzhovau, 2021-12-18:
Thanks for report. Looks like a bug in `alternatives` opkg option.

ryzhovau, 2021-12-21:
Fixed with https://github.com/Entware/ipk-html-indexer/commit/ad38e2061ff88d57bad53e67502b03ac54094cfe

closed 2021-12-21 by ryzhovau